dhylands/rshell #1: udev not found

dastultz, 2016-01-10:
Hello, this looks like a pretty cool tool. I followed the instructions to install but get this when I run:

```
black-widow-3:~ daryl$ rshell -p /dev/tty.usbmodem1412
Connecting to /dev/tty.usbmodem1412 ...
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.4/bin/rshell", line 9, in <module>
    load_entry_point('rshell==0.0.1', 'console_scripts', 'rshell')()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/rshell/main.py", line 2234, in main
    autoconnect()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/rshell/main.py", line 178, in autoconnect
    context = pyudev.Context()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/pyudev/core.py", line 95, in __init__
    self._libudev = load_udev_library()
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/pyudev/_libudev.py", line 270, in load_udev_library
    raise ImportError('No library named udev')
ImportError: No library named udev
Exception ignored in: <bound method Context.__del__ of <pyudev.core.Context object at 0x103172828>>
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.4/lib/python3.4/site-packages/pyudev/core.py", line 99, in __del__
    self._libudev.udev_unref(self)
AttributeError: 'Context' object has no attribute '_libudev'
```

I'm running on OSX 10.11.2. Am I doing something wrong? Is this a Mac issue?

Thanks!

dhylands, 2016-01-10:
Hmm. So I've really only done testing using linux. I did add some hooks so that it would work without pyudev.

Since you're calling it named rshell, I'm assuming that you used pip to install it right? It looks like pyudev got installed even though udev doesn't exist for the Mac. I'll have to see if I can fix that as a dependency.

If you're comfortable editing the source, you could try commenting out the call to autoconnect in main.py

That's the only place in the code that uses pyudev. I'll see if I can make some improvements so that autoconnect does some checking to see if its running under linux.

dastultz, 2016-01-10:
Yes, I used pip3 to install it. I commented out line 2234 and it does work now. It seems a little slow taking 4-5 seconds to do `ls /sd`. Still if I'm doing data logging and the SD card is not mounted this is a good alternative to copy files. (I normally mount the SD and use rsync to copy files, then pyboard.py to run my program.)

Thanks. I'm happy to do any further Mac-specific testing.

closed 2016-01-10 by dastultz

dhylands, 2016-01-10:
Thanks for letting me know. I'll get it fixed properly.

I have a mac here for testing, but it's a really old one, so it doesn't get used much.

There is a change coming down the pipe which should speed up the usb-serial transfer rates, which will in turn speed up things like ls -l. The code already sends a listdir request and the code running on the pyboard returns a list of files as well as doing an os.stat on each file (to determine file sizes, types, etc). Having a large number of files would definitely slow things down.

Feel free to let me know of any quirks. There are still lots of little things to improve.