dhylands/rshell #16: No connection possible if REPL does not come up after reset

ast0815, 2016-08-30:
Hello,

I just managed to put my first working python program on my ESP board using rshell (yay!).
Unfortunately now I cannot use rshell any more (boo!):

The problem seems to be that rshell resets the board, which then starts up the program in main.py instead of the REPL. Even if I abort the program by hand, with screen and CTRL-C, rshell resets the board which leads to the main.py to be loaded again. I am pretty much locked out now.

dhylands, 2016-08-31:
So yeah - rshell is built ontop of pyboard.py which uses the raw repl, which does a soft reset for each and every command that it executes.

I normally use rshell to get to the REPL and then import my script to run. If I want to be able to use rshell afterwards (once I've got it into main.py), then I normally use a GPIO connected to a switch or jumper to have it in "run" mode or in "repl" mode. If main.py exits quickly (what I'm calling repl mode), then rhsell should work well.

ast0815, 2016-08-31:
Wouldn't it be possible to send a CTRL-C after each reset (and some delay)?

dhylands, 2016-08-31:
pyboard.py already sends a couple of Control-C's.

I know on stmhal that Control-C's only work over the usb-serial and not over the UART. I wonder if that's related?

ast0815, 2016-09-01:
I'm using CTRL-C via screen to reset the board, so it should work. Maybe it's a timing issue?