dhylands/rshell #17: Not Windows Compatible

webtop, 2016-11-23:
This cannot be used out-of-the-box on Windows because the readline module is needed, which is not available for a Windows shell. Maybe you could detect OS and load a similar module for Windows users?

Thanks

dhylands, 2016-11-24:
It does work on windows under cygwin.

Are you aware of a similar module for windows?

I've managed to find pyreadline: https://pypi.python.org/pypi/pyreadline and I found this blog post on using it with rlcompleter:
http://www.farmckon.net/2009/08/rlcompleter-how-do-i-get-it-to-work/

Would you be able to test this?

Arrowana, 2017-06-01:
I tried to had some minor tweaks to make the rshell work on windows.
It does, I can use rshell. Absolute windows path don't seem to work though, but relative ones work fine
However pyreadline breaks the CPython REPL.

With pyreadline installed if I press TAB in the CPython REPL it will throw
`>>> def test():
... Readline internal error
Traceback (most recent call last):
  File "C:\Users\Arowana\AppData\Local\Programs\Python\Python35\lib\site-packages\pyreadline\console\console.py", line 768, in hook_wrapper_23
    res = ensure_str(readline_hook(prompt))
  File "C:\Users\Arowana\AppData\Local\Programs\Python\Python35\lib\site-packages\pyreadline\rlmain.py", line 571, in readline
    self._readline_from_keyboard()
  File "C:\Users\Arowana\AppData\Local\Programs\Python\Python35\lib\site-packages\pyreadline\rlmain.py", line 536, in _readline_from_keyboard
    if self._readline_from_keyboard_poll():
  File "C:\Users\Arowana\AppData\Local\Programs\Python\Python35\lib\site-packages\pyreadline\rlmain.py", line 556, in _readline_from_keyboard_poll
    result = self.mode.process_keyevent(event.keyinfo)
  File "C:\Users\Arowana\AppData\Local\Programs\Python\Python35\lib\site-packages\pyreadline\modes\emacs.py", line 243, in process_keyevent
    r = self.process_keyevent_queue[-1](keyinfo)
  File "C:\Users\Arowana\AppData\Local\Programs\Python\Python35\lib\site-packages\pyreadline\modes\emacs.py", line 286, in _process_keyevent
    r = dispatch_func(keyinfo)
  File "C:\Users\Arowana\AppData\Local\Programs\Python\Python35\lib\site-packages\pyreadline\modes\basemode.py", line 257, in complete
    completions = self._get_completions()
  File "C:\Users\Arowana\AppData\Local\Programs\Python\Python35\lib\site-packages\pyreadline\modes\basemode.py", line 200, in _get_completions
    r = self.completer(ensure_unicode(text), i)
  File "C:\Users\Arowana\AppData\Local\Programs\Python\Python35\lib\rlcompleter.py", line 80, in complete
    readline.redisplay()
AttributeError: module 'readline' has no attribute 'redisplay'
  File "<stdin>", line 2`

Obviously a pyreadline bug

ccooper21, 2017-12-04:
I just learned about `rshell` on the uPy forums and thought I'd give it a try (on Windows).  Pretty cool!  As expected, I had to install the `pyreadline` library to get it working.  Other than that, the only change I had to make is to replace the `'libedit' in readline.__doc__` reference in `main.py` with `readline.__doc__ and 'libedit' in readline.__doc__`.  I did not observe the error that @Arrowana reported.

Jim43, 2018-05-04:
@ccooper21 it works,thanks

RSabet, 2018-09-08:
see #62

clifforddw, 2018-10-08:
Hi All, I managed to get rshell working on windows through the Linux Subsystem for windows. 
install rshell as per the documentation and then enable the requires serial port with this command :

`stty -F /dev/ttyS2 -a`

Good article here with the background [https://blogs.msdn.microsoft.com/wsl/2017/04/14/serial-support-on-the-windows-subsystem-for-linux/](url)

karelv, 2018-12-20:
@dhylands , 

> I've managed to find pyreadline: https://pypi.python.org/pypi/pyreadline and I found this blog post on using it with rlcompleter:
> http://www.farmckon.net/2009/08/rlcompleter-how-do-i-get-it-to-work/

I did this:
`pip install rshell`
(rshell-0.0.16)
I run rshell, I got:

> ....
> ModuleNotFoundError: No module named 'readline'

Then I've installed pyreadline:
`pip install pyreadline`
(pyreadline-2.1)

```
$ rshell
Welcome to rshell. Use the exit command to exit rshell.

No MicroPython boards connected - use the connect command to add one

C:\Users\karelv\micropython

```
Note: I used `git-sh` as shell and python from `anaconda` on my win7 pc.
Note2: the prompt is in green, so I guess the colors are working
Note3: I entered repl, but no clue how to exit it... exit() is not working...

dhylands, 2018-12-20:
You should be able to exit the REPL by using Control-X. I didn't want to use Control-D since Control-D does a soft reset under MicroPython. It should print a line mentioning Control-X when you enter the REPL.

karelv, 2018-12-21:
Hello @dhylands, ctrl-X indeed exits the repl, while ctrl-D exits the rshell (when in git-sh).
Thanks a lot. Now I have another question, off-topic. I'm new to micropython and ESP32. I just now have a ESP32 module. I want to create an application with CAN bus 2.0, but I have misread the micropython doc, CAN is only implemented for pyboard. What would you suggest me to do? Go arduino, or go ESP-IDF as development platform? Is it possible to 'port' arduino to micropython? Or I go better with ESP-IDF?

karelv, 2018-12-21:
Again ontopic... I tried rshell also on cmd.exe (win7).
* `help` seems a built-in function in DOS
* `exit` is also a built-in function in DOS, so it exits not only rshell, but also the cmd.exe.
* now indeed Ctrl-D is not working.

dhylands, 2018-12-21:
I have'n't really used ESP-IDF directly, and I rarely use arduino, so I'm not really sure which would be more appropriate.

As for help and exit being built-in commands to cmd.exe, I don't understand the issue. Those commands will only be interpreted by cmd.exe if you're at the cmd prompt. If you're at the rshell prompt, then the commands will be interpreted by rshell.

So yeah, if you type the command twice, the first one will be interpreted by rshell and when it exits, the second buffered command will be executed by cmd.exe.

webtop, 2018-12-21:
I personally use ESP-IDF, simply because it is developed by Espressif, so
there is never going to be any versioning issues with different ESP
modules. While I do love Arduino, it is so much more open to issues because
of it's community focus.

The ESP-IDF site has great documentation, and if you're looking for a
community to help you, check out esp32.com