dhylands/rshell #18: b key not going through

askvictor, 2016-12-08:
Seems like a PEBCAK (or perhaps its a practical joke?) but when I press the 'b' key into rshell, it doesn't do anything.  All other keys work fine, and b works fine everywhere else. On OSX, running rshell 0.0.6. I'm testing with an esp8266 board, but it does this even without a connection to the board.

dhylands, 2016-12-08:
Sounds very strange. It seems to be working fine for me.
```
1997 >rshell
Welcome to rshell. Use Control-D to exit.

No MicroPython boards connected - use the connect command to add one

/home/dhylands> boards
No boards connected
/home/dhylands> 
```
I'm away from home this week, but I'll try it on my Mac when I get home.

yanjost, 2017-01-11:
I can confirm, have the same issue

rshell version 0.0.8
tested with tmux or without
under iTerm and Terminal

Maybe related to readline ?

What can I do to help ?

dhylands, 2017-01-11:
Did you disconnect any other terminal software connected to the pyboard? (just making sure because I've seen this cause all kinds of weird behaviour).

I just tried 0.0.8 on my Mac Mini (which is running OS X Yosemite - 10.10.5)
```
dhylands@dave-mini ~
504 >rshell --version
0.0.8
dhylands@dave-mini ~ 
505 >rshell
Connecting to /dev/cu.usbmodem642 ...
Welcome to rshell. Use Control-D to exit.
repl
Entering REPL. Use Control-X to exit.
>
MicroPython v1.8.6-205-g91359c8-dirty on 2016-12-19; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
>>> print('abc')
abc
>>> 
```
that was ssh'ing into the mini.

Does using the --nocolor option make things behave any differently?

rshell uses python's Cmd module, so it doesn't even process the input directly. cmd.py calls sys.stdin.readline() to get its input. So this script:
```
import sys
print('Enter some text')
line = sys.stdin.readline()
print("Read '" + line + "'")
```
should be doing the same thing as rshell.

I tested using Terminal as well and everything seems to be working fine. 

Do you get the same results when using the virtual keyboard (http://osxdaily.com/2013/06/21/mac-virtual-keyboard-os-x/)

dhylands, 2017-01-11:
When you launch rshell, there are 2 modes. The first is command mode where you type stuff like the repl command and commands to copy files. Then there is the repl mode.

Does the b key not work in both modes? Or just one? The repl mode doesn't use Cmd's input mechanism, so if it was isolated to one mode or the other that might provide useful information.

TravisJoe, 2017-01-12:
I have had strange things happen depending on hardware and which terminal app I am using. Screen with Mac and Linux seems to be the most solid.

dhylands, 2017-01-12:
> I have had strange things happen depending on hardware and which terminal app I am using. Screen with Mac and Linux seems to be the most solid.

In this case, rshell is the terminal program, and any other teminal programs (just as screen) need to be disconnected from the pyboard.

gunderson, 2017-03-11:
Same issue when using Terminal in OS X 10.11.6 on Sparkfun's ESP32 Thing board

braffour, 2017-03-25:
Same for me, on OSX el capitan, running rshell with robotdyn uno esp8266 (ch340 usb driver), not able to connect (no problem with screen)

With my wemos mini D1 (ch340 driver), I'm able to connect, all commands work fine except b key

AVanVlack, 2017-03-28:
Having this issue as well, thing I noticed on my system (Mac, Sierra)

-Its only lowercase 'b', Uppercase works fine.
-It's giving feedback via error sounds when pressed.
-Happens once I enter rshell, connected to a board or not. But only in command mode, once I enter shell (shell command) or repl (repl command) lowercase 'b' works fine.

Such an odd issue.

gr4viton, 2017-03-28:
Oh,
now it gets me (i don't have this issue). But, I am using '\b' printed in python on Windows 7 to generate the bell sound (was pc speaker in the older days, now it's a windows sound).
So isn't it this kind of problem? rshell prepresenting the lower be as the \b char ?
(it's from a group of \n and \t control chars)
but now i see it's '\a' for bell. and \b for backspace.. so maybe not..
https://en.wikipedia.org/wiki/Control_character

dhylands, 2017-03-28:
It sounds like somebody thinks that the control key is being pressed. rshell doesn't actually do any input processing itself when in command mode, that's all done by Cmd (a library that comes with python).

When you get into this situation, does pressing and release either left and/or right control keys a few times change anything? i.e. just press and release the left control key a few times and then press the b, and then repeat with the right control key (I'm assuming that the mac keyboards have a control key).

askvictor, 2017-03-28:
I don't know if this is related, but on pyqt (and qt in general) on Mac,  the control modifier flag refers to the command key, and vice versa.

dhylands, 2017-05-01:
I just got a new MacBook Pro and was able to reproduce the problem.

The problem is related to readline. Older versions of python/OSX used libedit. Newer versions use the actual GNU readline. The libedit command to enable tab completion caused the b key to get bound incorrectly when run with readline.

I committed a fix that should work on old and new versions.

dhylands, 2017-05-01:
Fixed in 5f05d8b870ea91d02d33937701ac629c02dea401

closed 2017-05-01 by dhylands