dhylands/rshell #20: Problem writing to /flash via UART

peterhinch, 2017-01-19:
Setup: a Pyboard V1.1 connected via an FTDI adaptor to UART2. No USB connection. Rshell invocation:
rshell -p /dev/ftdi --editor nano --buffer-size=30
boot.py:
```python
import pyb
pyb.usb_mode(None)
uart = pyb.UART(2, 115200)
pyb.repl_uart(uart)
```
At the rshell prompt I can copy files of all sizes to /sd
Copying to /flash is unreliable. Small (< 300 bytes) files sometimes work, but sometimes fail to replace existing files. A 19.1K file appears to copy (the red LED comes on for a while) but the rshell prompt never reappears and the Pyboard seems to have crashed: a reboot is required to reconnect.

dhylands, 2017-01-21:
I can reproduce the problem. I'm pretty sure that what's happening is that interrupts are being disabled for a period of time while the flash is being written and characters are getting dropped.

Now to see if I can figure out a way to work around it.

dhylands, 2017-01-21:
It seems that while a flash erase is happening, if an attempt is made to access flash (to fetch CPU instructions, read a constant or any other reason) then the CPU stalls until the flash erase occurs.

My current speculation is that this is happening and UART characters are being dropped. So I need to see if I can arrange some type of flow control to get the host to stop sending data while the flash is being manipulated.

Pmaene, 2017-03-02:
The file operations of Adafruit's ampy tool rely on Python code executed on the board (https://github.com/adafruit/ampy/blob/master/ampy/files.py#L125). This might be an alternative to the UART-based approach?