dhylands/rshell #24: Can't access /flash

pacmac, 2017-02-17:
I have rshell installed and connected to my esp8266 via UART, however I am unable to access the file system, I can access the device using the repl command, but all of the filesystem commends only appear to work on the local file system ??

    ls -l /flash
    Cannot access '/flash': No such file or directory

If I access via repl:
   >>> os.listdir()
  ['boot.py', 'webrepl_cfg.py', 'net.py', 'flash']

I thought that maybe the device did not have a flash folder, so just in case I created one off the root, but the error is the same, it does not appear to access the remote fs at all??

What am I doing wrong ??

dhylands, 2017-02-17:
The esp8266 doesn't have /flash. It makes the internal flash filesystem be at root /

However, rshell considers / to be on the host filesystem. There is an alternative in that you can use /pyboard to reference the root filesystem of the first connected board.

pacmac, 2017-02-17:
Thanks, that works but I now have another issue, it is verrrrrry slow.

I am connected at -b 115200 and it takes around 5 seconds to connect, once connected if I enter ls /pyboard, only around 5 seconds later is the listing produced.

I don't have any speed issues when connected normally, any ideas ??

dhylands, 2017-02-17:
Every "command" that rshell sends to the board requires pyboard.py to execute it, which in turns causes a soft-reset.

At start time, rshell sends 4 commands to the board. The ls command will send 2 commands.

So if your boot.py or main.py command has any delays in it, then that will slow down rshell.

You should see the delay by pressing Control-D from the repl. Any delay seen doing a Control-D will be seen by rshell.