dhylands/rshell #27: ESP32 "Could not enter raw repl"

markserrano915, 2017-06-22:
This is actually an issue and a response I posted on the ampy project. See [ESP8266 Micropython "Could not enter raw repl" #19](https://github.com/adafruit/ampy/issues/19)

I am creating this issue because it's exactly the same with rshell. This is for reference and workaround purposes. 

I had been trying to solve the same problem with my ESP32 for two days now (I think I got the same issue on ESP8266 01 but I probably gave up). But then out of the blue I was able to make it work today.

So when I plugin the board on my Macbook 2013, that's when I get this unable to enter REPL issue. But as soon as I plug it in on my Raspberry Pi, I am able to run the same rshell commands without issues!

My immediate solution is:

1. Do the coding in Mac (because it's more convenient to type in there). 
2. Push to git
3. Pull in the code to my Raspberry Pi
4. Use Raspberry Pi's for uploading code to the board via rshell

Some interesting observations:
* When I use picocom on my Mac, before I am able to enter on the Micropython REPL shell, the terminal spits out some garbage text and afterwards I can work on picocom without any issues (until I use ampy or rshell)
* When I use picocom on my Raspberry, there are no pre-garbage text. I go straight to the REPL with no issues with ampy and rshell

dhylands, 2017-06-22:
Are you disconnecting picocom before trying to use rshell? If not - that's part of the problem. You can't have 2 programs both trying to use the same usb-serial port. If you try, one program will get some of the characters and the other program will get some of the characters.

markserrano915, 2017-06-23:
@dhylands Nope. I don't use picocom and rshell together. It's either I am at picocom terminal or trying ampy / rshell after I have exited out picocom. 

The usb cable is connected directly to the MacBook's USB port. I am not using any usb hub. So Macbook has two USB ports. The other one is using a USB hub so I can use other devices, ie iphone, hard drives etc.

nmistry, 2018-02-26:
Seems like this issue still persists... Will try to upgrade to the latest MicroPython, but this is what i get on a brand new wemos esp32 Lolin32 lite dev board, after freshly installing rshell: 

`(uPython) ➜  ~ rshell -b 115200 -d --buffer-size=30 -p /dev/tty.wchusbserial1420
Debug = True
Port = /dev/tty.wchusbserial1420
Baud = 115200
User = micro
Password = python
Wait = 0
nocolor = 0
binary = 0
Timing = 0
Quiet = 0
Buffer_size = 30
Cmd = []
Connecting to /dev/tty.wchusbserial1420 ...
----- About to send 286 bytes of code to the pyboard -----
def test_buffer():
    """Checks the micropython firmware to see if sys.stdin.buffer exists."""
    import sys
    try:
        _ = sys.stdin.buffer
        return True
    except:
        return False
output = test_buffer()
if output is None:
    print("None")
else:
    print(output)

-----
b'\xd9:\xc2A0\x8c\xdcE\xc9v\xba\xc2A0I\xfdd.\x97\x82\xc2\xc1\xb0i\xcd\xc1_\xe4\xae\x97\x82\xc2\xc10,\xa8\xd6E\xc9v\xba\xc2A0i\xc1_\xa4\xae\x97\x82\xc2\xc10\r\xa1\xeb\x8b\x95:\xc4R\x8a\t\x1a\xb1\xbdc\xeb"\xa5v\'fH\xec+\xd1:\xc2\xcdf\xe6\x16\x821bb\x95\xb9i\xb4C!\xeb\x85d\xba\xc2\xcd\xe6\x96A0L,\x89\x95n\xbaS\x9a2\x8d!\xeb\x85\xd1:\xc2\xd10\xb0\x13\x820bb\x95\xb9:\xb0C!\xeb\x85d\xba\xc2Q\xb0\xba\xc20\xb0b\xa2\x95\xf9:L\xca9\xb2C\xa1\x96\x8b\xc9\xe5 \xc2\xd10\xb0\x13YcSH\xf8\x1b[0;32mI (349) cpu_start: Pro cpu up.\x1b[0m\r\n\x1b[0;32mI (349) cpu_start: Single core mode\x1b[0m\r\n\x1b[0;32mI (350) heap_init: Initializing. RAM available for dynamic allocation:\x1b[0m\r\n\x1b[0;32mI (353) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (359) heap_init: At 3FFD4D08 len 0000B2F8 (44 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (366) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM\x1b[0m\r\n\x1b[0;32mI (372) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM\x1b[0m\r\n\x1b[0;32mI (378) heap_init: At 4008FB00 len 00010500 (65 KiB): IRAM\x1b[0m\r\n\x1b[0;32mI (384) cpu_start: Pro cpu start user code\x1b[0m\r\n\x1b[0;32mI (178) cpu_start: Starting scheduler on PRO CPU.\x1b[0m\r\nOSError: [Errno 2] ENOENT\r\nMicroPython v1.9.2-283-gdba3796e on 2017-10-27; ESP32 module with ESP32\r\nType "help()" for more information.\r\n>>> '
Traceback (most recent call last):
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/main.py", line 1185, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno 8] nodename nor servname provided, or not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/nmistry/.virtualenvs/uPython/bin/rshell", line 11, in <module>
    sys.exit(main())
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/main.py", line 2658, in main
    real_main()
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/main.py", line 2620, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/main.py", line 1191, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/main.py", line 1215, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/main.py", line 1436, in __init__
    Device.__init__(self, pyb)
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/main.py", line 1262, in __init__
    self.has_buffer = self.remote_eval(test_buffer)
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/main.py", line 1353, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/main.py", line 1331, in remote
    self.pyb.enter_raw_repl()
  File "/Users/nmistry/.virtualenvs/uPython/lib/python3.6/site-packages/rshell/pyboard.py", line 187, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
rshell.pyboard.PyboardError: could not enter raw repl
(uPython) ➜  ~`

dhylands, 2018-02-26:
Thanks for the report. I never appreciated all the crap being sent back
during boot up.

I noticed this:
OSError: [Errno 2] ENOENT

Which is probably coming from your boot.py. I suspect that you'll need to
fix that first before rshell will work properly.

Once you've got that dealt with you may also need to add the -a option in
order to transfer files.

bachya, 2018-03-20:
I'm having the same issue as the OP: a brand new Wemos D1 Mini flashed with MicroPython 1.9.3 shows the same garbage before getting to the REPL:

```
rll��|�l�|�d�c|����{�c�c��no�lgg���cp��l{drlp�o��dbo�|�ldc��gg�d��l`�nnd�l`'s���o�sd``�o��c�ol���no��dx�n�r�����c'�|$�c��oo�l�d`�gnd`gsǓ�orl p�o�s�������bo�|��c��og�d`�gd`o;���g�`s��o�l �l��|��rg#��l�b�dc쌜���dc�drlll��|��{oc��ld�x{�c�#䄜��b#��$sl�l��|��rgc��d���c�b�Ĝ��b�lcd����o�r��g|�dlll`��s�d�l�d`��{�l�l�l`��s�d���ll`rl��;d����c��c�ccs�{c��'�o�l��l�l��dd������l�o����cdd�rp���cl�c;lrls�o�����o����c������l#lsc��㌇l{d��do�psd�d��|����#4 ets_task(40100130, 3, 3fff837c, 4)
OSError: [Errno 2] ENOENT

MicroPython v1.9.3-8-g63826ac5c on 2017-11-01; ESP module with ESP8266
Type "help()" for more information.
>>>
```

As discussed, this means I can't use `rshell` – I eventually get the `could not enter raw repl` error.

@dhylands, re: your comment about this being an issue in `boot.py`, this occurs even with the default `boot.py` that MicroPython uses:

```
# This file is executed on every boot (including wake-boot from deepsleep)
#import esp
#esp.osdebug(None)
import gc
#import webrepl
#webrepl.start()
gc.collect()
```

Not sure what to fix (or whether anything can be "fixed" at this level)?

Sansanito, 2018-08-29:
My apologies, I mixed the themes. I have the same problem: "Could not enter raw repl" but when I use ampy from cmd on Windows. Anyway, I suppose that in the case of rshell, in the case of using python3 there is also an "egg" package with a "pyboard.py" file to edit, or it will be free in the "site-packages" folder, such as in python2

patatchi, 2018-09-26:
> The only thing that worked for me was changing the file "pyboard.py" in Python_path / Lib / site-packages /
> (for Python3, first change the extension to the adafruit_ampy-X.X.X-py.X.X file from .egg to .zip and unzip, edit the file and then change the extension again).
> In the "pyboard.py" file, you have to add a pause in the definition of the "enter_raw_repl" function, after the line
> self.serial.write (b'\r\x03\x03')
> add
> time.sleep (.2)
> 

This works!

I have the same issue with using ampy with my NodeMCU ESP8266, got the "could  not enter raw repl" error on all my windows 10 machines at work. It works fine on OSX. But adding this line made it work! Thanks Sansanito!!

hyper99, 2018-10-03:
I like rshell very much. but i meet the same issue with ESP32- WROOM-32. I must hardware reset the ESP32, then can connect to ESP32 each time, but on ESP8266, It's ok.  
 I add below code in "pyboard.py" file, the function  just simulates the hareware reset before  connection. 
                       self.serial = serial.Serial(device, baudrate=baudrate, interCharTimeout=1)
                #--------added  below---------------------
               self.serial.setDTR(0) #IO0=HIGH
               time.sleep(0.02)
               self.serial.setRTS(1) #EN=LOW
               time.sleep(1)
               self.serial.setDTR(1) #IO0=LOW
               time.sleep(0.02)
               self.serial.setRTS(0) #EN=HIGH
               time.sleep(1)
               self.serial.setDTR(0) #IO0=HIGH
               #------ added to here--------
               break
The other delay time in the code sometimes just makes ESP32 work, and ESP8266 does not.
I adjusted the delay time, and it work on ESP32 and ESP8266 DEVKITV1 both。
Maybe the delay time depends on the capacity  on EN pin and IO0 pin. 

sorry for my poor english.

hyper99, 2018-10-04:
simplified to below code  can reset ESP board when connecting,  and no need hardware reset when reconnecting to ESP board  on ESP8266 and ESP32 board.
in __init__() of class pyboard.py file.  
        self.serial = serial.Serial(device, baudrate=baudrate, interCharTimeout=1)
#--------added  below--------------------
self.serial.setRTS(1) #EN=LOW
time.sleep(0.01)
self.serial.setRTS(0) #EN=HIGH
#------ added to here--------
break

nichtleiter, 2018-10-26:
Version: 0.0.16

On my system rshell also works very unreliable.

After I wrote a simple blink Test to the board `rshell` puts the following on the screen

```
johannes:~/iot/esp/Micropython/MyExamples/TimerTest % rshell -b 115200 -p /dev/ttyUSB0
Connecting to /dev/ttyUSB0 ...
b'ets Jun  8 2016 00:22:57\r\n\r\nrst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)\r\nconfigsip: 0, SPIWP:0xee\r\nclk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\nmode:DIO, clock div:1\r\nload:0x3fff0018,len:4\r\nload:0x3fff001c,len:4732\r\nload:0x40078000,len:7496\r\nload:0x40080400,len:5512\r\nentry 0x4008114c\r\n\x1b[0;33mW (86) boot: PRO CPU has been reset by WDT.\x1b[0m\r\n\x1b[0;33mW (86) boot: WDT reset info: PRO CPU PC=0x400834d4\x1b[0m\r\n\x1b[0;33mW (86) boot: WDT reset info: APP CPU PC=0xf6dd7dfb\x1b[0m\r\n\x1b[0;32mI (402) cpu_start: Pro cpu up.\x1b[0m\r\n\x1b[0;32mI (402) cpu_start: Single core mode\x1b[0m\r\n\x1b[0;32mI (402) heap_init: Initializing. RAM available for dynamic allocation:\x1b[0m\r\n\x1b[0;32mI (406) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (412) heap_init: At 3FFC4F68 len 0001B098 (108 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (418) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM\x1b[0m\r\n\x1b[0;32mI (424) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM\x1b[0m\r\n\x1b[0;32mI (431) heap_init: At 40091604 len 0000E9FC (58 KiB): IRAM\x1b[0m\r\n\x1b[0;32mI (437) cpu_start: Pro cpu start user code\x1b[0m\r\n\x1b[0;32mI (7) cpu_start: Starting scheduler on PRO CPU.\x1b[0m\r\nalloc_emergency_exception_buf done\r\nLet\'s start blinking\r\nMicroPython v1.9.4-674-g27ca9ab8b on 2018-10-24; ESP32 module with ESP32\r\nType "help()" for more information.\r\n>>> '
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1197, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/rshell", line 11, in <module>
    load_entry_point('rshell==0.0.16', 'console_scripts', 'rshell')()
  File "/usr/local/lib/python3.6/dist-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 2730, in main
    real_main()
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 2692, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1203, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1227, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1473, in __init__
    Device.__init__(self, pyb)
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1285, in __init__
    self.root_dirs = ['/{}/'.format(dir) for dir in self.remote_eval(listdir, '/')]
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1390, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1368, in remote
    self.pyb.enter_raw_repl()
  File "/usr/local/lib/python3.6/dist-packages/rshell/pyboard.py", line 193, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
rshell.pyboard.PyboardError: could not enter raw repl
```

The same does `ampy`.

At the same time `mpfshell` works without complaining. Here a quick sample session i started right after invoking `rshell`:

```
johannes:~/iot/esp/Micropython/MyExamples/TimerTest % mpfshell -c "open ttyUSB0"
Connected to esp32

** Micropython File Shell v0.8.1, sw@kaltpost.de **
-- Running on Python 2.7 using PySerial 3.4 --

mpfs [/]> cat boot.py
# This file is executed on every boot (including wake-boot from deepsleep)
#import esp
#esp.osdebug(None)
#import webrepl
#webrepl.start()

mpfs [/]> cat main.py
import machine
from machine import Pin
from machine import Timer
import micropython

micropython.alloc_emergency_exception_buf(256)

print("alloc_emergency_exception_buf done")

io23 = Pin(23, Pin.OUT)
io05 = Pin(5, Pin.OPEN_DRAIN)

io23.value(1)
io05.value(0)

timer1 = Timer(-1)
timer1.init(mode=Timer.PERIODIC, period=100, callback=lambda t:io23.value(not io23.value()))

timer2 = Timer(-2)
timer2.init(mode=Timer.PERIODIC, period=100, callback=lambda t:io05.value(not io05.value()))

print("Let's start blinking")

mpfs [/]>
```

ropg, 2021-05-24:
I had the same problem. When looking into it I found that the ili9341 driver I use does not deinit cleanly, which causes the following behaviour on Ctrl-D (soft reset):

```text
Traceback (most recent call last):
  File "ili9XXX.py", line 244, in deinit
MemoryError: memory allocation failed, heap is locked
MPY: soft reboot
Single buffer
assertion "ATB_GET_KIND(block) == AT_HEAD" failed: file "../../py/gc.c", line 591, function: gc_free
abort() was called at PC 0x401ea00f on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40099224:0x3ffd4590 0x4009964d:0x3ffd45b0 0x401ea00f:0x3ffd45d0 0x400d34ee:0x3ffd4600 0x400d2bcd:0x3ffd4620 0x40106eaf:0x3ffd4640 0x400ffdc2:0x3ffd4660 0x4012cd55:0x3ffd4680 0x4012dbf3:0x3ffd46a0 0x40100925:0x3ffd46d0 0x4015e223:0x3ffd4700 0x400e44d5:0x3ffd4720 0x400e02f1:0x3ffd4740 0x400e0432:0x3ffd4760 0x400e5e80:0x3ffd4780 0x400e5ebb:0x3ffd47b0 0x400eb978:0x3ffd47d0 0x400e04ba:0x3ffd48b0 0x400eedb5:0x3ffd48f0 0x400e46a4:0x3ffd4990 0x400e02f1:0x3ffd49c0 0x400e0432:0x3ffd49e0 0x400eeaa1:0x3ffd4a00 0x400e46a4:0x3ffd4aa0 0x400e02f1:0x3ffd4ad0 0x400e0432:0x3ffd4af0 0x400eeaa1:0x3ffd4b10 0x400e46a4:0x3ffd4bb0 0x400e02f1:0x3ffd4be0 0x400e2c32:0x3ffd4c00 0x400e02f1:0x3ffd4c50 0x400e0432:0x3ffd4c70 0x400ea46e:0x3ffd4c90 0x400e9919:0x3ffd4ce0 0x400e02f1:0x3ffd4d00 0x400ee9e1:0x3ffd4d20 0x400e46a4:0x3ffd4dc0 0x400e02f1:0x3ffd4df0 0x400e031a:0x3ffd4e10 0x4015dc7e:0x3ffd4e30 0x4015e061:0x3ffd4ec0 0x4015e0cb:0x3ffd4ee0 0x400f8587:0x3ffd4f00 0x40095d25:0x3ffd4f30

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5092
load:0x40078000,len:14024
load:0x40080400,len:3760
entry 0x40080638
Double buffer
ILI9341 initialization completed
FT6X36 touch IC ready (fw id 0x10 rel 1, lib 300E)
MicroPython v1.14-457-g3bf6c7f12-dirty on 2021-05-23; M5Stack Core2 with ESP32
Type "help()" for more information.
>>> 
```

The problem is that rshell, via pyboard, does the following when it goes to the raw REPL mode:

* ctrl-C twice
* flush input
* Ctrl-A
* wait for `raw REPL; CTRL-B to exit\r\n>`
* ctrl-D
* wait for `raw REPL; CTRL-B to exit\r\n`

Apparently this soft-reset maintains the raw REPL status and does not revert to friendly REPL. But since my ESP32 crashes on the soft reset because the ili9341 driver doesn't cleanly unload, it doesn't get a raw REPL when it expects one. You'll notice the second time it looks for the raw REPL thing, it looks without the `>` prompt. That is significant because that is then expected by `exec_raw_no_follow` (line 230), so just removing the last two steps doesn't work.

The soft reset is probably to unload anything that might be hogging memory. One might argue that this is a broken state on the board and that therefore rshell doesn't need to work, but this will lock out everyone that gets their board in a state where the soft-reset does not happen cleanly, although they have a perfectly working raw REPL.

So I built a command line option `-s`, `--suppress-reset` to both pyboard and rshell. It does what you might expect and lets users connect to boards that are not exhibiting the correct soft-reboot behaviour. See the [Pull Request](https://github.com/dhylands/rshell/pull/161) for details.

ropg, 2021-05-24:
If you want to try if `-s` solves your problems:

`python -m pip install git+https://github.com/ropg/rshell.git`