dhylands/rshell #29: download "cp" to ESP32 hangs

fluffymcpuffin, 2017-08-17:
Hi,
on a linux x64 (KDE Plasma) Kubuntu system:
i can not download to my ESP32 devkit V1 from doit.
the filename is written to /pyboard  correctly.
had to us ampy to download and the use rshell.

just for you to know...
thanks
juergen

dhylands, 2017-08-18:
The typical reason for this is having a buffer-size which is too big. Unfortunately, the way rshell currently copies files isn't very robust, and dropping a single character is enough to make things hang.

So if the UART on the upy board can't keep up with continuous bursts of buffer-size characters without dropping a character, then you need to keep dropping buffer-size until it does work.

On the ESP8266, a buffer-size around 30 seems to work for most people. I've not worked with the ESP32 yet, so I don't have a feel for how well the UART code is, and how much latency the interrupt handlers have.

fluffymcpuffin, 2017-08-19:
thank's
--buffer-size=32 does work ok...
the uart bridge uses a CP2102

one other observation though: FYI
when i exit out of repl with ^X and exit out of rshell with ^D all is well.
BUT: when i go back in with rshell  --> see below
it is not a host usb-uart issue.
if i hit the reset on the esp32 all is well. 
so the ^X (exiting the esp32) leaves it in a different state as a hard reset.
==========================================================
jb@i5-6600k:~/Desktop/Downloads/micropython-bmp180-master$ rshell --buffer-size=32
Connecting to /dev/ttyUSB0 ...
b'\x03\x03 \x07\xe7\xbb\xe4'
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1185, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/rshell", line 11, in <module>
    load_entry_point('rshell==0.0.10', 'console_scripts', 'rshell')()
  File "/usr/local/lib/python3.5/dist-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 2646, in main
    real_main()
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 2608, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1191, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1215, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1435, in __init__
    Device.__init__(self, pyb)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1261, in __init__
    self.has_buffer = self.remote_eval(test_buffer)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1352, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1330, in remote
    self.pyb.enter_raw_repl()
  File "/usr/local/lib/python3.5/dist-packages/rshell/pyboard.py", line 187, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
rshell.pyboard.PyboardError: could not enter raw repl

dhylands, 2017-08-19:
Control-X is basically just an exit to the host side of the REPL. It doesn't send anything to the pyboard (since you may be running a program and you don't want to interrupt it). Control-D is then just an exit to rshell.

When you relaunch rshell, it will try to send some commands to the board. rshell relies on pyboard.py to interrupt the currently runnning program by sending Control-C followed by some control characters to enter the raw-repl.

If I had to guess, I would say that the ESP32 isn't responding to the Control-C's in order to interrupt the currently running program.

So yes, exiting rshell does not leave the board in the same state as a hard reset, and that's by design. It leaves it with whatever program was running, running.

fluffymcpuffin, 2017-08-19:
thank's
it may be different if rshell would issue a ^D (soft reset of ESP32)
if the control characters of the ESP32 are different than those of the pyboard that would explain it.
(see attached below)

i had no program running when leaving the esp back to the rshell and back do bash.
going back and forth from within rshell to repl and back again works just fine.

just need to remember to hit the reset on the ESP before going to rshell.
thank you...
jb

>>> help()
Welcome to MicroPython on the ESP32!
Control commands:
  CTRL-A        -- on a blank line, enter raw REPL mode
  CTRL-B        -- on a blank line, enter normal REPL mode
  CTRL-C        -- interrupt a running program
  CTRL-D        -- on a blank line, do a soft reset of the board
  CTRL-E        -- on a blank line, enter paste mode

fluffymcpuffin, 2017-08-20:
all is well using an ESP8266
the ESP32 has above problem....

closed 2017-08-20 by fluffymcpuffin

twhtanghk, 2022-05-14:
experienced the same proble. Finally, successfully cp to pyboard with buffer size set to 1 as below
"rshell -b 115200 -p /dev/ttyUSB0 --buffer-size 1"