dhylands/rshell #33: Connection just hangs with LOBO ESP32WROOM

diginfo, 2017-11-22:
Hi Dave;

Have been using AMPY until now but are trying rshell due to rsync but can't seem to get it to work.

See below, as you can see initially I list the local dir and then attempt rsync to the ESP32WROVER, yes the apps folder does exist on the WROVER.

1. Any idea what the problem may be ??

2. Any chance of adding the ability to add the folder if it does not exist as this would mean that I could create a dist(ribution) folder and run rsync and have it make an exact copy of the dist folder. Using current design I would need to manually add the target folders first ?!

```
PACBOOK:flash pac$ rshell -p $slab
Connecting to /dev/tty.SLAB_USBtoUART ...
Welcome to rshell. Use Control-D to exit.
/Volumes/Data/uPython/flash> ls
apps/           cert/           device/         html/           lib/            picoweb-master/ sys/            gv.py          
boot/           dev/            fonts/          json/           main/           reg/            test/           main.py        
/Volumes/Data/uPython/flash> rsync apps /flash/apps
Traceback (most recent call last):
  File "<stdin>", line 52, in <module>
  File "<stdin>", line 47, in recv_file_from_host
  File "<stdin>", line 34, in recv_file_from_host
KeyboardInterrupt: 
>OKTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'T' is not defined
>OKTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'high' is not defined
>OKTraceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>raw REPL; CTRL-B to exit
>OKTraceback (most recent call last):
  File "<stdin>", line 2
SyntaxError: invalid syntax
>OKTraceback (most recent call last):
  File "<stdin>", line 2
SyntaxError: invalid syntax
>OKTraceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```

diginfo, 2017-11-22:
I get the same errors with cp -r

dhylands, 2017-11-27:
Try starting rshell with the -a option.

closed 2019-03-31 by diginfo