dhylands/rshell #37: rshell won't connect to micropython on esp8266

larsks, 2017-11-28:


larsks, 2017-11-28:
Ah, I've solved this one.  It's because I had a file named `board.py`, which apparently conflicts with something that `rshell` was expecting?  There is no `board` module in the esp8266 port, at least.  I guess I will need to find another name.

closed 2017-11-28 by larsks

dhylands, 2017-11-28:
Ah yes - rshell supports connecting to multiple boards and it does this by looking for a boards.py file.

cefn, 2018-06-03:
Is there a workaround which doesn't involve renaming board.py?

I have an impossible situation since there has to be a module named exactly board.py for compatibility with Circuitpython...
https://github.com/adafruit/Adafruit_Micropython_Blinka/tree/master/src

I ran into this when trying to rsync packages to a Micropython board with the Blinka compatibility layer installed and don't know what options I have, (aside from scripting ampy to remove and replace the file after every rsync).

cefn, 2018-06-04:
As a workaround I just made sure there was in fact a string name value within Blinka board.py. What principled way is there to populate this? Currently it's populated according to the board type, but that will mean a clash when using rshell with e.g. more than one Feather Huzzah. Is there a minimal identifier already available in memory somewhere (e.g. a unique identifier which doesn't require loading e.g. a network library to get a mac address).

dhylands, 2018-06-04:
You could edit main.py in rshell here:
https://github.com/dhylands/rshell/blob/b148481f2f4fd4c567a151ac4a905e118a0b9857/rshell/main.py#L527-L534
Change 'import board' and 'board.name' to something else.

I'll need to figure out an alternative to work with CircuitPython