dhylands/rshell #39: rshell not finding flash file system

slzatz, 2017-12-26:
I am using the @loboris micropython ESP32 port on an Adafruit Feather ESP32 and am running rshell on ubuntu 17.10.  I start rshell with:

    rshell --port /dev/ttyUSB0 --buffer-size=32

I can get to the repl through rshell with no problem

When I use the boards cmd I get:

    pyboard @ /dev.ttyUSB0 connected Dirs: /flash /pyboard/flash

but

    ls /flash

results in `No such file or directory` and

    ls /pyboard

results in this traceback.

     File "../rshell/bin/rshell", line 11, in <module>
        load_entry_point('rshell==0.0.11', 'console_scripts', 'rshell')()
     File "../rshell/command_line.py", line 4, in main
        rshell.main.main()
     File "../rshell/main.py", line 2658, in main
         real_main()
     File "../rshell/main.py", line 2641, in real_main
        shell.cmdloop(cmd_line)
     File "/../rshell/main.py", line 1551, in cmdloop
        cmd.Cmd.cmdloop(self)
     File "/usr/lib/python3.6/cmd.py", line 138, in cmdloop
        stop = self.onecmd(line)
     File "/../rshell/main.py", line 1581, in onecmd
        self.onecmd_exec("".join(group))
     File "/../rshell/main.py", line 1592, in onecmd_exec
        return cmd.Cmd.onecmd(self, line)
     File "/usr/lib/python3.6/cmd.py", line 217, in onecmd
        return func(arg)
     File "../rshell/main.py", line 2169, in do_ls
        ldir_stat = auto(listdir_stat, filename)
     File "/../rshell/main.py", line 519, in auto
        return dev.remote_eval(func, dev_filename, *args, **kwargs)
     File "..rshell/main.py", line 1353, in remote_eval
        return eval(self.remote(func, *args, **kwargs))
      File "<string>", line 0
       ^
Note that:

    ampy -p /dev/ttyUSB0 ls /flash

works fine with the same board and host set-up.

Dave, any thoughts?

dhylands, 2017-12-27:
Not sure. I haven't yet tried rshell on the ESP32. Typically these are related to file system issues. When you do an `ls /flash` command, rshell tries to execute `os.stat('/')` followed by `os.listdir('/')` followed by os.stat on each of the files returned by os.listdir.

It would be good to know what output you get when running those commands manually from the REPL.

If you run rshell with the -d option it will print out a bunch more info, including the commands being executed. It would be useful to find out exactly where its failing.

slzatz, 2017-12-27:
Dave, I used -d and it mirrored what you're going to see below taken directly from the REPL.   Not sure why `os.stat("/flash")` generates an exception.

    >>> import os
    >>> os.stat("/")
    (16384, 0, 0, 0, 0, 0, 0, 0, 0, 0)

    >>> os.stat("/flash")
    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
    OSError: [Errno 22] EINVAL
   
    >>> os.listdir("/")
    ['flash']
    >>> os.listdir("/flash")
    ['boot.py', 'iot_esp32_tft.py', 'topic', 'config.py', 'mqtt_id', 'main.py']

dhylands, 2017-12-28:
Thanks. That's a bug in the filesystem code, that was fixed quite a while ago, and the ESP32 fork probably doesn't have it.

I'll see about adding some code to rshell which causes it to deal with that error.

slzatz, 2017-12-28:
Dave, thanks for taking a look.  I am using the @loboris ESP32 version and not sure whether the issue is fixed in the upstream micropython ESP32 port or not.

slzatz, 2017-12-30:
I updated the firmware to:

    MicroPython ESP32_LoBo_v2.0.8 - 2017-11-04 on ESP32 board with ESP32

and everything works fine so the issue was not testing with the latest version.

closed 2017-12-30 by slzatz