dhylands/rshell #40: RSHELL  failed to access /dev/ttyUSB0

Novo55, 2018-01-05:
don't understand why RSHELL can't see my ESP32 when Picocom can.

rshell is version 0.0.11

Running Ububtu 16.04 on dell I7
[image]

[image]

Novo55, 2018-01-05:
when I use sudo on the command it works, I'm such a newbie :-)

closed 2018-01-05 by Novo55

dhylands, 2018-01-05:
There are two ways to make it so that you don't need to use sudo (and this is just a linux thing, nothing to do with rshell):
1 - Add your user to the dialout group. If you do `ls -l /dev/ttyUSB0` you'll typically see something like this:
```
crw-rw---- 1 root dialout 188, 0 Dec 18 20:54 /dev/ttyUSB0
```
This says that the user root and the group dialout have read/write access to /dev/ttyUSB0. Note, when adding yourself to a group, you'll need to logout completely and log back in for it to take effect. You can use the `id` command to verify what groups your user is a member of.

2 - You can add a file to /etc/udev/rules.d which will change the permissions of the /dev/ttyUSB0 file. Follow the directions here: https://github.com/micropython/micropython/wiki/Board-STM32F407-Discovery#programming-from-linux-via-dfu (note that you don't need to install dfu-utils)

Novo55, 2018-01-20:
Hi Dave,

been using RSHELL for a couple of weeks its an awesome tool, please keep 
up the good work.

regards

-- 
regards Steve W

tejastank, 2020-03-20:
# I face same issue.

sudo chmod 777 /dev/ttyUSB0

Above command works for me  ;)

davehylands, 2020-03-20:
I added some notes to the rshell README: https://github.com/dhylands/rshell#serial-port-permissions-linux