dhylands/rshell #41: Serial from URL is not supported

kleckie, 2018-02-26:
Pyserial supports the use of a URL for the port address. This is useful for utilizing ports over the network. In my case using an RFC2217 port. 

Please consider changing serial.Serial() to serial.serial_for_url() in the file  pyboard.py. 

The change seems to work for me.

dhylands, 2018-02-27:
pyboard.py actually comes from the micropython project:
https://github.com/micropython/micropython/blob/master/tools/pyboard.py

I periodically sync up this version with that version. So the best way would be to submit a PR for the micropython project.

Otherwise, submitting a PR here could also work, and I can try to get it merged upstream. I'd prefer to get a PR or a diff (rather than just a description of the change as in comment 1), just because I don't currently have any devices I can test with.

kleckie, 2018-02-27:
Thanks for your reply. I will close issue and look into the Micropython change.

closed 2018-02-27 by kleckie