dhylands/rshell #42: Problem in file edit

huberthoegl, 2018-04-15:
Hello, 

I am trying to edit a file with "edit /flash/hh.py" in rshell, but an exception is thrown.  The traceback is

```
`Traceback` (most recent call last):
  File "/home/hhoegl/anaconda3/bin/rshell", line 11, in <module>
    load_entry_point('rshell==0.0.11', 'console_scripts', 'rshell')()
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/main.py", line 2659, in main
    real_main()
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/main.py", line 2642, in real_main
    shell.cmdloop(cmd_line)
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/main.py", line 1552, in cmdloop
    cmd.Cmd.cmdloop(self)
  File "/home/hhoegl/anaconda3/lib/python3.6/cmd.py", line 138, in cmdloop
    stop = self.onecmd(line)
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/main.py", line 1582, in onecmd
    self.onecmd_exec("".join(group))
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/main.py", line 1593, in onecmd_exec
    return cmd.Cmd.onecmd(self, line)
  File "/home/hhoegl/anaconda3/lib/python3.6/cmd.py", line 217, in onecmd
    return func(arg)
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/main.py", line 2038, in do_edit
    cp(filename, local_filename)
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/main.py", line 584, in cp
    filesize, xfer_func=recv_file_from_remote)
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/main.py", line 1336, in remote
    xfer_func(self, *args, **kwargs)
  File "/home/hhoegl/anaconda3/lib/python3.6/site-packages/rshell-0.0.11-py3.6.egg/rshell/main.py", line 987, in recv_file_from_remote
    dst_file.write(binascii.unhexlify(write_buf[0:read_size]))
binascii.Error: Non-hexadecimal digit found
```

I already tried some debugging and found that in main.py, line 986 

    dst_file.write(binascii.unhexlify(write_buf[0:read_size]))    

write_buf contains the following data on the last iteration, which is the reason for the exception:

b'632829290a202020207072696e74282264657669636520455549203d222c206d616329\xf0ets Jun  8 2016 00:22:57\r\n\r\nrst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)\r\nconfigsip: 0, SPIWP:0xee\r\nclk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\nmode:DIO, clock div:1\r\nload:0x3fff8028,len:8\r\nload:0x3fff8030,len:1728\r\nload:0x4009fa00,len:0\r\nload:0x4009fa00,len:14584\r\nentry 0x400a059c\r\nMicroPython v1756e28293a0a202020206d6163203d2062696e61736369692e6865786c696679286e6574776f726b2e4c6f526128292e6d61'

I am using a LoPy1.0r with up-to-date firmware. On the host computer I have Python 3.6.4 (Anaconda). 

With kind regards,

Hubert

huberthoegl, 2018-04-15:
Solved!

A few minutes after posting I found out, that I have to set the buffer size to a larger value than the
default value 512, e.g.

   rshell -p /dev/ttyUSB0 --buffer-size 2048

Now everything works well. Thanks for the nice rshell tool!

dhylands, 2018-04-15:
Normally, you need to decrease the buffer size to a value like 30. This is especially true for the devices which don't have native USB support. The pyboard supports USB natively, most of the others require the use of a USB-to-serial converter, which is often built into the board.

huberthoegl, 2018-04-15:
I tried it now with a buffer-size of 30, but execution of "edit" aborts again as I described it above. The problem has something to do with the file size. My file hh.py is 571 byte long. Smaller files with only a few Python source lines always can be edited without any problem for the default buffer size 512.  My LoPy [1] UART is driven by a FT234XD which sits on the LoPy expansion board V2 [2].

[1] https://pycom.io/product/lopy/
[2] https://pycom.io/product/expansion-board-2-0/