dhylands/rshell #43: rshell copying file to ESP8285 corrupts filesystem

cefn, 2018-04-20:
I had unreliable behaviour from rshell on ESP8285, while esptool, ampy and GNU screen were able to manipulate the device (suggesting that the serial drivers and the board itself are OK). 

The board is running Micropython 1.9.3 and I installed rshell through pip3

After attempting to connect, then run `cp color.py /pyboard/color.py` I am forced to CTRL+C to restore liveness since otherwise the terminal remains hung with no response.

After starting with a freshly flashed ESP8286, I tried to copy then killed the hung session with KeyboardInterrupt. I then checked to identify the board by running `boards`. It then apparently reported a huge number of connected directories and filesystems.

Here is the first console session which hangs (note the line overwriting)...

$ rshell
Welcome to rshell. Use Control-D to exit.

No MicroPython boards connected - use the connect command to add one

v/ttyUSB0n/Documents/shrimping/git/cockle/modules> connect serial /de 
Connecting to /dev/ttyUSB0 ...
rdome/cefn/Documents/shrimping/git/cockle/modules> cp color.py /pyboa 

Here is the second console session...

$ rshell
Welcome to rshell. Use Control-D to exit.

No MicroPython boards connected - use the connect command to add one
/home/cefn/Documents/shrimping/git/cockle/modules> boards
No boards connected
/home/cefn/Documents/shrimping/git/cockle/modules> connect serial /dev/ttyUSB0
Connecting to /dev/ttyUSB0 ...
/home/cefn/Documents/shrimping/git/cockle/modules> boards
pyboard @ /dev/ttyUSB0 connected Dirs: /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/. /pyboard/.
/home/cefn/Documents/shrimping/git/cockle/modules>

Invoking `ps aux | grep rshell` and `ps aux | grep screen` lists no zombie rshells or GNU screen sessions still sitting around unkilled. 

On closer inspection through GNU screen, from a newly-flashed board, simply connecting and running 

```bash
cp color.py /pyboard/color.py 
```

...seems enough to fully corrupt the filesystem. After running that single rshell command I get the following on my board...

```python

>>> import os
>>> os.listdir()
['\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x0 [... cut ...] 00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00', '\x00\x00\x00\x00\x00\x00\x00\x00.\x00\x00\x00']
>>> 
```

Running with `rshell --ascii` also leads to a hang, but force restarting the board only shows a zero-byte file called color.py so filesystem isn't corrupted in that case.

dhylands, 2018-04-20:
When using rshell with an ESP8266, you need to specify --buffer-size=30 otherwise it will use its default of 512, which only works for microcontrollers which natively support USB (the ESP8266 doesn't).

cefn, 2018-04-20:
Fixed, thanks for the extra guidance!

closed 2018-04-20 by cefn