dhylands/rshell #46: Feature Request mv / rename

diginfo, 2018-05-13:
Would be good to have this command too:

`mv pyboard/boot.py pyboard/boot.px`

dhylands, 2018-05-14:
I agree. I've wanted to use that several times recently myself.

rhubarbdog, 2019-04-19:
@dhylands have you worked on this or is it something I could do for you and offer a pull request.
I love rshell nice one

dhylands, 2019-04-19:
Feel free to submit a PR. I've already addressed your other two issues that you raised (better error for baod = RED) and improved ls in regards to symlinks. You'll find it in the latest master.

There is a script called `r.py` which I use to run rshell directly from the git repository.

peterhinch, 2019-09-03:
If my proposed [macro facility](https://github.com/dhylands/rshell/issues/111) is implemented, this can be done with the macro
```python
macros['mv'] = 'cp {0} {1}; rm {0}'
```