dhylands/rshell #47: Feature Request - Command Chaining & Non-Volatile History

diginfo, 2018-05-16:
Sorry for all of the feature requests, but have been using rshell a lot over the last few weeks and these would be really nice and frequently used features, for me anyway:

**1) Command Chaining:**
`cp /somefile.py esp32/lib; cp /anotherfile.py esp32/dev; repl`

**2) Non-Volatile History**
sometimes rshell crashes or I have to exit for some other reason, and if I have been copying files with some long paths (my memory is not so good) it can take me a minute or so to figure out the paths I was using before the quit.

Saving this history for the last x most recent commands would be very helpful when resuming working after restarting rshell again.

dhylands, 2018-05-17:
You can also prime the repl by using ~ So you can do:
```
repl ~ import somefile
```

diginfo, 2018-05-18:
This is a very nice feature, I use it all the time.

FYI I much prefer rshell over using FTP

jose1711, 2022-09-01:
When it comes to history that would survive across sessions - I just used an example from `readline` documentation at https://docs.python.org/3/library/readline.html#example only replacing `python_history` with `rshell_history`. I would raise a PR but it looks like the maintenance of this project sadly has come to a halt.