dhylands/rshell #53: pyboard - False when copying files

diginfo, 2018-06-22:
I am encountering problems on all 3 of my MAC's whereby larger files (<20k) fail to upload.

I am using SD card so there's no problem with storage.

The file transfer starts, the green LED lights up, then a few seconds later it goes out and False is reported in the console.

After this I have to disconnect / reboot the pyboard.

I have tried playing with buffer size from 64 - 1024, tried enabling disabling ascii transfer mode, no difference.

I have also disabled usb storage mode to prevent that from interfering.

diginfo, 2018-06-22:
I just upgraded to 0.14, now there is no 'false' message, however the transfer still fails.

dhylands, 2018-06-22:
What board are you connecting to? For ESP8266/ESP32 I normally use buffer-size=32 and I also have to use the -a flag to force ASCII-only transfers.

diginfo, 2018-06-22:
> pyboard - False when copying files

as per the title PYBOARD, yes I have already tried all of those options:

> I have tried playing with buffer size from 64 - 1024, tried enabling disabling ascii transfer mode, no difference.

closed 2019-03-31 by diginfo