dhylands/rshell #60: rshell not working in win10

Easydo1, 2018-07-08:
Hi,

I've tryed to install rshell but it dit not work. Here is wat happend. I've opend cmd.exe and executed te commands i've marked in **bold** for readability.

C:\Users\Ivo\AppData\Local\Programs\Python\Python37\Scripts>**pip3 install rshell**
Collecting rshell
  Downloading https://files.pythonhosted.org/packages/14/e1/cb99b0a5d4ad33c7033d8800d6a12c857189a92604ddf1087be45c6653a4/rshell-0.0.14.tar.gz
Requirement already satisfied: pyserial in c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages (from rshell) (3.4)
Collecting pyudev>=0.16 (from rshell)
  Downloading https://files.pythonhosted.org/packages/bc/a2/31a07829acea8e70a28c247f43fa5d981229ae0f9edfeddedf52de00709b/pyudev-0.21.0.tar.gz (89kB)
    100% |████████████████████████████████| 92kB 1.5MB/s
Collecting six (from pyudev>=0.16->rshell)
  Downloading https://files.pythonhosted.org/packages/67/4b/141a581104b1f6397bfa78ac9d43d8ad29a7ca43ea90a2d863fe3056e86a/six-1.11.0-py2.py3-none-any.whl
Installing collected packages: six, pyudev, rshell
  Running setup.py install for pyudev ... done
  Running setup.py install for rshell ... done
Successfully installed pyudev-0.21.0 rshell-0.0.14 six-1.11.0

C:\Users\Ivo\AppData\Local\Programs\Python\Python37\Scripts>**rshell**
Traceback (most recent call last):
  File "C:\Users\Ivo\AppData\Local\Programs\Python\Python37\Scripts\rshell-script.py", line 11, in <module>
    load_entry_point('rshell==0.0.14', 'console_scripts', 'rshell')()
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\pkg_resources\__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\pkg_resources\__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\pkg_resources\__init__.py", line 2324, in load
    return self.resolve()
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\pkg_resources\__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\rshell\command_line.py", line 1, in <module>
    import rshell.main
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\rshell\main.py", line 65, in <module>
    import readline
ModuleNotFoundError: No module named 'readline'

C:\Users\Ivo\AppData\Local\Programs\Python\Python37\Scripts>**rshell -h**
Traceback (most recent call last):
  File "C:\Users\Ivo\AppData\Local\Programs\Python\Python37\Scripts\rshell-script.py", line 11, in <module>
    load_entry_point('rshell==0.0.14', 'console_scripts', 'rshell')()
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\pkg_resources\__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\pkg_resources\__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\pkg_resources\__init__.py", line 2324, in load
    return self.resolve()
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\pkg_resources\__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\rshell\command_line.py", line 1, in <module>
    import rshell.main
  File "c:\users\ivo\appdata\local\programs\python\python37\lib\site-packages\rshell\main.py", line 65, in <module>
    import readline
ModuleNotFoundError: No module named 'readline'

C:\Users\Ivo\AppData\Local\Programs\Python\Python37\Scripts>

Jim43, 2018-07-29:
may be you you should install a python module named 'readline',try`pip3 install readline`?

lukpaz, 2018-08-29:
I've got the same error:
```
PS C:\Users\user\source\repos\micropython-ide-vscode> pip install rshell
Requirement already satisfied: rshell in c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages (0.0.14)
Requirement already satisfied: pyserial in c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages (from rshell) (3.4)
Requirement already satisfied: pyudev>=0.16 in c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages (from rshell) (0.21.0)
Requirement already satisfied: six in c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages (from pyudev>=0.16->rshell) (1.11.0)
PS C:\Users\user\source\repos\micropython-ide-vscode> rshell
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python37-32\Scripts\rshell-script.py", line 11, in <module>
    load_entry_point('rshell==0.0.14', 'console_scripts', 'rshell')()
  File "c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages\pkg_resources\__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages\pkg_resources\__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages\pkg_resources\__init__.py", line 2324, in load
    return self.resolve()
  File "c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages\pkg_resources\__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages\rshell\command_line.py", line 1, in <module>
    import rshell.main
  File "c:\users\user\appdata\local\programs\python\python37-32\lib\site-packages\rshell\main.py", line 65, in <module>
    import readline
ModuleNotFoundError: No module named 'readline'
PS C:\Users\user\source\repos\micropython-ide-vscode> pip install readline
Collecting readline
  Downloading https://files.pythonhosted.org/packages/f4/01/2cf081af8d880b44939a5f1b446551a7f8d59eae414277fd0c303757ff1b/readline-6.2.4.1.tar.gz (2.3MB)
    100% |████████████████████████████████| 2.3MB 2.6MB/s
    Complete output from command python setup.py egg_info:
    error: this module is not meant to work on Windows

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\user\AppData\Local\Temp\pip-install-0cjbee0y\readline\
```
It seems that readline is not meant to work on Windows so rshell won't work.
Any workarounds?

yeyeto2788, 2018-09-17:
@dhylands, @Easydo1 I have the same error on Windows 7.

@Jim43  this is actually the return of try installing the module (Not a windows module).

```
$ pip3 install readline
Collecting readline
  Downloading https://files.pythonhosted.org/packages/f4/01/2cf081af8d880b44939a5f1b446551a7f8d59eae414277fd0c303757ff1b/readline-6.2.4.1.tar.gz (2.3MB)
    Complete output from command python setup.py egg_info:
    error: this module is not meant to work on Windows
```

Nanzikambe, 2018-10-04:
Same issue here as @Easydo1 on Windows 10, and also unable to install readline:

> error: this module is not meant to work on Windows

mattytrentini, 2018-10-18:
I'm having the same issues on Windows 8. 

`pyreadline` is a common replacement for `readline` on Windows and provides much of the same functionality. It also adds a drop-in `readline` module intended to replace the `readline` module. This gets past the `import readline` issue but fails on the following line:

```python
if 'libedit' in readline.__doc__:
```
Since apparently `pyreadline` doesn't have a docstring. Commenting out lines 67-70 allows the application to work using `pyreadline`:

```python
#if 'libedit' in readline.__doc__:
#    readline.parse_and_bind ("bind ^I rl_complete")
#    BROKEN_READLINE = True
# else:
readline.parse_and_bind("tab: complete")
```

`rshell` works under Windows for me after this change (including tab completion).

In short; `readline` is _not_ supported under Windows. `pyreadline` is intended to be a drop-in replacement for `readline` on Windows that _could_ be used but a work-around needs to be found to support the 'BROKEN_READLINE' fix. Alternatively that commented-out section could be selectively executed only if `readline` was being used.

Fundamentally a better cross-platform solution for `readline` is really needed...

mattytrentini, 2018-10-19:
Also suggest changing the ticket name to "rshell not working on Windows" since it's not specific to Windows 10.

mattytrentini, 2018-10-20:
That's great work @dhylands, thanks! Reviewed the changes and they look good. I'll test the changes at work next week when I'm back in front of a Windows box.

haaslukas, 2018-11-29:
`pip3 install readline` worked for me

gregorypruden, 2019-01-30:
`pip3 install pyreadline` worked for me on Windows 10 build

mzdaniel, 2019-02-02:
Encounter the same issue and just submitted #74 for this.

Elamarang1, 2024-10-25:
Interrupting a currently running program to stop execution when new code is uploaded.
Using rshell, these methods are functioning correctly. However, when a loop-based program is already running on the board, I encounter issues uploading new code. For example, when the following program is running:

python

Copy code
print("New petal.py running...")

# Example main program loop

while True:
    print("Hello Elamaran from the NEW petal.py!")
The new code doesn't upload or execute as expected.

Could you advise on how to handle this issue?