dhylands/rshell #61: Logo proposal for RShell

mchobby, 2018-09-01:
Hi Dave,
I'm using RShell a lot and really love it. I even created a French documentation for it ;-)
[French RShell Tutorial](https://wiki.mchobby.be/index.php?title=MicroPython-Hack-RShell)
In that documentation, I did drawed a logo (using snake skin in background).
I would like to offer it to the RShell project... feel free to use it.
[RShell Logo on the French doc](https://wiki.mchobby.be/images/4/4b/RShell-logo.jpg)

Kind regards,
Dominique

dhylands, 2018-09-03:
Excellent work on the translation. If you don't mind I'll post a link to it from the README.

The only thing I really noticed is that you have my name as "Daniel Hylands" when it should read as "Dave Hylands".

mchobby, 2018-09-05:
Hi Dave,
I did fix your name. I do appologies.
For sure, you can link the French tutorial.
Have a nice day.
Dominique