dhylands/rshell #65: Unable to connect to NRF port device

diginfo, 2018-10-28:
I have been working on the NRF port and when I attempt to connect to the nrf52840, I get the errors below.

I am able to connect to this device using linux / osx 'screen' and are also able to connect rshell to other devices such as pyboard, so it looks like this is specific to the nrf port.

Could this be something to do with the filesystem on the nrf port ??

```
Connecting to /dev/tty.SLAB_USBtoUART ...
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rshell/main.py", line 1185, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno 8] nodename nor servname provided, or not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/rshell", line 11, in <module>
    load_entry_point('rshell==0.0.11', 'console_scripts', 'rshell')()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rshell/main.py", line 2658, in main
    real_main()
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rshell/main.py", line 2620, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rshell/main.py", line 1191, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rshell/main.py", line 1215, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rshell/main.py", line 1436, in __init__
    Device.__init__(self, pyb)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rshell/main.py", line 1266, in __init__
    elif not self.remote_eval(test_unhexlify):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/rshell/main.py", line 1353, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "<string>", line 0
    
    ^
SyntaxError: unexpected EOF while parsing
```

dhylands, 2018-10-29:
What command line are you using to start rshell?

If you connect via screen what does the output of these commands show?
```
>>> import os
>>> os.listdir('/')
```

dhylands, 2018-10-29:
Also, what is the output of these commands:
```
>>> import ubinascii
>>> ubinascii.unhexlify
```

pacmac, 2018-10-30:
```
>>> import os
import os
>>> os.listdir('/')
os.listdir('/')
Traceback (most recent call last):
  File "<stdin>", in <module>
TypeError: function takes 0 positional arguments but 1 were given
```

```
>>> os.listdir()
os.listdir()
[]
```

```
>>> import ubinascii
import ubinascii
Traceback (most recent call last):
  File "<stdin>", in <module>
ImportError: no module named 'ubinascii'
```

dhylands, 2018-10-30:
I'll guess that rshell is crashing on the fact that ubinascii is missing. I should be able to fix that. However, rshell won't be of much use to you until os.listsir('/') actually returns something. Until it does, rshell can't copy anything since it's basically saying that there are no filesystems on the device.

dhylands, 2018-10-31:
Yeah - I think that the micro filesystem isn't really intended to be changed, and probably doesn't support subdirectories.

dhylands, 2018-10-31:
You should be able to build it with sdcard support though, and those function may magically appear in that case. Then we just need to make sure os.listdir('/') works when using an sdcard.

pacmac, 2018-10-31:
I have enabled:

`MICROPY_FATFS=1`

In the build command, but os and uos still don't have the additional commands and os.listdir() will not work with a variable i.e os.listdir('/')

So I suspect that this build option may not be fully working.

How would I enable SD support for the build, either in the GNUmakefile or in the build command as I can't find any hints on that ?

dhylands, 2018-10-31:
I saw this in the nrf port main.c: https://github.com/micropython/micropython/blob/06643a0df4e85cbdf18549440db19dc21fccbf76/ports/nrf/main.c#L162 but I don't see any other references to MICROPY_HW_HAS_SDCARD in the nrf port tree. There are generic drivers to support sdcard over SPI elsewhere in the micropython tree.

jose1711, 2022-08-30:
> Yeah - I think that the micro filesystem isn't really intended to be changed, and probably doesn't support subdirectories.

Well.. that depends on the definition of a filesystem change. It still can create and remove files. This is from a test on my microbit flashed with MicroPython:
```
>>> dir(os)
['__name__', 'remove', 'listdir', 'size', 'uname']
>>> f = open('test', 'w')
>>> f.write('xxx')
3
>>> f.close()
>>> open('test').read()  
'xxx'
>>> os.listdir()
['main.py', 'test']
>>> os.remove('test')
>>> os.listdir()     
['main.py']
>>> open('test', 'a') 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: illegal mode
```
That means:
  * no support for directories (looks like it does not have any clue about *any* directory, even the root one)
  * files can be created, read/written and deleted
  * file appending is not supported