dhylands/rshell #66: fix for MAC for could not enter raw_repl bug

philwilkinson40, 2018-11-14:
when using MAC to access RSHELL the following error
could not enter raw_repl

rough fix in 
pyboard.py
do you want me to push this ?

using the platform module so only triggered on MAC
insert  line 182
```
if platform.platform()[:6] == "Darwin":
    time.sleep(2)
```

dhylands, 2018-11-15:
When you said insert line 182, did you mean inside the while loop or not?

philwilkinson40, 2018-11-16:
No, sorry I wasn't very clear. Outside the while loop.

philwilkinson40, 2018-11-16:
I have tested this with Linux and MAC on esp8266 and ESP32 and all connect fine with the small addition. I realise it is a bit of a hack!!!

dhylands, 2018-11-16:
Was Micropython running something or at the REPL when you run into this problem?

I really don't want to add the delay to every command sent, but adding it to say the first one I don't have a problem with. Starting Micropython winds up sending 4-5 different commands to the board, which would wind up adding 10 seconds of delay.