dhylands/rshell #67: ESP32: unable to connect to rshell the second time

rprr, 2018-12-02:
I am having a strange issue with an ESP32. After a power cycle, I can connect via rshell to /dev/ttyUSB0. If I quit rshell and try to start again, it does not connect. The /dev/ttyUSB0 device is there but does not connect. If I power cycle it, the connection takes place easily the first time. Not sure what is wrong.

dhylands, 2018-12-02:
How are you exiting rshell?
When you exited rshell, was the board still at the REPL prompt? Or was it running a program?
What command line options are you using when you start rshell?
If you exit rshell can you connect with another terminal program? (i.e. putty, picocom, etc)]
When you start rshell the second time, could you try adding the -d option and reporting back what gets output?

rprr, 2018-12-02:
I usually exit rshell via ctrl-d or exit. The problem happens even if I don't enter repl. 

### First Time:

$ sudo rshell -a -d --buffer-size=32 -p /dev/ttyUSB0 
Debug = True
Port = /dev/ttyUSB0
Baud = 115200
User = micro
Password = python
Wait = 0
nocolor = 0
binary = 1
Timing = 0
Quiet = 0
Buffer_size = 32
Cmd = []
Connecting to /dev/ttyUSB0 ...
----- About to send 304 bytes of code to the pyboard -----
def test_unhexlify():
    """Checks the micropython firmware to see if ubinascii.unhexlify exists."""
    import ubinascii
    try:
        _ = ubinascii.unhexlify
        return True
    except:
        return False
output = test_unhexlify()
if output is None:
    print("None")
else:
    print(output)

-----

-----Response-----
b'True\r\n'
-----
MicroPython has unhexlify
----- About to send 222 bytes of code to the pyboard -----
def listdir(dirname):
    """Returns a list of filenames contained in the named directory."""
    import os
    return os.listdir(dirname)
output = listdir('/')
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b"['flash']\r\n"
-----
----- About to send 386 bytes of code to the pyboard -----
def set_time(rtc_time):
    rtc = None
    try:
        import pyb
        rtc = pyb.RTC()
        rtc.datetime(rtc_time)
    except:
        try:
            import machine
            rtc = machine.RTC()
            rtc.datetime(rtc_time)
        except:
            pass
output = set_time((2018, 12, 2, 7, 12, 33, 35, 0))
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b'None\r\n'
-----
----- About to send 292 bytes of code to the pyboard -----
def board_name(default):
    """Returns the boards name (if available)."""
    try:
        import board
        name = board.name
    except ImportError:
        name = default
    return repr(name)
output = board_name('pyboard')
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b"'pyboard'\r\n"
-----
----- About to send 334 bytes of code to the pyboard -----
def get_time_epoch():
    """Determines the epoch used by the MicroPython board."""
    import time
    try:
      return time.gmtime(0)
    except:
      """Assume its a pyboard, with an epoch of 2000."""
      return (2000, 1, 1, 0, 0, 0, 0, 0)
output = get_time_epoch()
if output is None:
    print("None")
else:
    print(output)

-----
-----Response-----
b'(1970, 1, 1, 0, 0, 0, 5, 1)\r\n'
-----
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.
/home/ramp> 
Executing ""
/home/ramp> Executing "EOF"

### Second Time:

(xenial)ramp@localhost:~$ sudo rshell -a -d --buffer-size=32 -p /dev/ttyUSB0 
Debug = True
Port = /dev/ttyUSB0
Baud = 115200
User = micro
Password = python
Wait = 0
nocolor = 0
binary = 1
Timing = 0
Quiet = 0
Buffer_size = 32
Cmd = []
Connecting to /dev/ttyUSB0 ...
----- About to send 304 bytes of code to the pyboard -----
def test_unhexlify():
    """Checks the micropython firmware to see if ubinascii.unhexlify exists."""
    import ubinascii
    try:
        _ = ubinascii.unhexlify
        return True
    except:
        return False
output = test_unhexlify()
if output is None:
    print("None")
else:
    print(output)

-----
b'\x04\xff'
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1197, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/rshell", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 2730, in main
    real_main()
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 2692, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1203, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1227, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1473, in __init__
    Device.__init__(self, pyb)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1280, in __init__
    elif not self.remote_eval(test_unhexlify):
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1390, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1368, in remote
    self.pyb.enter_raw_repl()
  File "/usr/local/lib/python3.5/dist-packages/rshell/pyboard.py", line 188, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
rshell.pyboard.PyboardError: could not enter raw repl

rprr, 2018-12-02:
When I try picocom, I don't get the repl prompt even after I type enter a few times.

dhylands, 2018-12-02:
Thanks for the report, I'll see if I can figure out what's happening.

rprr, 2018-12-02:
Ok, I tried on my Mac. No problems connecting and reconnecting. My problems were happening on 1) Raspberry Pi and 2) Crouton on a Chromebook. Both have arm processors and are running arm version of linux. Maybe there is an issue with the USB/Serial driver for the CP210X chipset.

rprr, 2018-12-03:
And now it is worked properly. I'm perplexed.  I have tried it a few times entering and exiting. 

Log of activities for today:

1. Install espruino firmware.
2. The first step did not work so I erased flash first
3. Reinstall espruino
4. Play with espruino a little. Notice esp32 was getting hot.
5. Erase flash
6. Reinstall loboris esp32 firmware
7. Now everything is working. 

Maybe the flash erasing did the trick this time. Puzzled. :)

Closing the issue for now.

closed 2018-12-03 by rprr

dhylands, 2018-12-10:
And also, are you using the loboris version of ESP32 Micropython or the one from the micropython repository?

rprr, 2018-12-11:
> And also, are you using the loboris version of ESP32 Micropython or the one from the micropython repository?

I have used both but had problems irrespective. Now things appear to be fixed. I now make it a plan to erase_flash before write_flash.

tmolteno, 2019-07-04:
Hi. I am experiencing this error connecting to a MaixPy. Exactly the same exception

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/rshell/main.py", line 1282, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:
```
I have managed to successfully connect once (randomly, after uninstalling and reinstalling rshell a few times and resetting the board). Remaining attempts have since failed.

dhylands, 2019-07-05:
I think that the error is rather misleading (and should be fixed). I think that the real cause is that rshell is unable to open the serial port, and then it falls back to thinking it should be a telnet connection.

One of the typical things that causes problems is when modemmanager is installed (under linux). It will open serial ports and send AT commands thinking that its a modem, and if rshell tries to connect while modemmanager has the port open then that causes problems.

Rather than unplugging and replugging the board, try just hitting the reset switch on the board in question.