dhylands/rshell #68: Failure to connect

chrisvelevitch, 2018-12-09:
I'm running `rshell` v0.0.14 on a Mac running OS X 10.13.6 and Python 3.7.0 and I'm trying to connect to a `D1 mini` which is `ESP-8266EX`-based from Wemos.cc, with a board rev. V3.1.0 and when I try to connect, I getting this error. Note: when I connect to a board which is rev. V3.0.0, I'm not getting this error.
```
$ rshell -p /dev/tty.wchusbserial14120
Connecting to /dev/tty.wchusbserial14120 ...
b';l\x00d\x9c\x9e|\x00\x8cl\xe0<\x03\x0c\x04\x0c\x84\x0cl\xe4\x0cb<\x87\x83\x03\xe4\x1b\x93{\x92c\x8c\x04c\x84\xfbgn\x9fdo\'\x9c\xe3\xe4\x04c\x1cp\xec\x8e${l{$x\xf3o\xe0\x18\x03\x0c\x04\x82\x04l\x04\x04\x0c\x04\x0c\x04c\x0co\xe2<\x03\xec\x84\x0c\x04\x0c\x0cc\x8c\xfbo\'\xe7\x00d\xc4\xc7$`\x03\xd8\x1b\x13ng\x04l \x03\x07\x03or\xc7\x9b\x92n\x0c\x04\xdb\x93\xdb`\x03p\xf2n\xe0\x18\x03\x0c\x04{\x8c\x9c\x9c\xe3\xe0\x0c\x04\x04\x0cb\x0c\'\xe3|\x03ld\x04\x0c\x8fc\x84\xf2\'o\xef\x00\x04\x0c$`\x03\x90\x12\x1bo\'\x04l`\x03\x07\x03gs\x8e\xdb\x93o\x04\x04\x13\x1b\x93`\x038\xfbg\xe0\x10\x03\x04\x04s\xc4\x9c\x9c\xe3\xe0\x04\x87s\x80\x04c\x04g\xe3|\x03\xe4\x04\x0c\x04\x0c#\x8c\xf3og\xe7\x00\x0c\x04l \x03\x98\x13\x1bgn\x0cd`\x03\x0e\x03g{\x8f\x93\x9bg\x04\x0cs\x03d`\x03\x07{\x92\x9bo\x04\x0cs\x03l`\x03\x07\x04\x04l\x9c\xdf|\xff\x82\x1b;soc\x8f\x8c\x04l\xe4\x1b\x0c\x04c\xe4l\x04c\xec\xc4\x9c\x9c\xe3\xe3\x04\xc4\x0c\x04c\x04\x87lrd\x07l\x04l\x9c\x9f<\xff\x83\x1bs;gb\x87\x84$l\x9c\x1c\x83\x04\x04c\xe4\x8c\x04c\xec\xc4\x9c\x9c\xe3\x03\x13\x1b\x02\x04c\xe4\x04\x87dsd\x0e\xc4\x0cd\x9c\x9e|\xff\x83\x1br{oc\x8f\x8cd\x0c\xdc\x1c\x93\x0c\x04c\x8c\x04\x0cb\xec\x84\x9c\xdc\xe3\x03\x02\x84\x0c\x0cb\xecdc\x0f$\x8c\xe2\x03\xec\x13\xc3g\xe4\x92{\xc3\xf3o<\xe4\x04l\x04$ld`\x03\x8c\xe3s\xdbl\x84l\x1b\xc4\x0c\x04\x0cd`\x02\xc4\xe3r\x9b$\xecd\x13\x84\x04\x0c\x04l \x03\x8c\xe2{\x93d\x04\x87\x9f\x00\x8c\x0c\x04ld`\x02;l\x87\x83{d\x84\x9f\xe3\xc4c\x0c\x84\xc4c\x84c\x04cs\x03\x03\xec\x8e;\x04\x13\x1bc\xe4\xfbg\xe7\x80\x12\'o\xe3\x18\x03\x0c\x03$\x92\xfb\x03\x0c$\x0c\x84\x04l\x84\x87d$\xc7\x1f\x83\x8c\x84\xc4\xec\x8f\x07d\x80\x03g\xfc\x00\x8c\x9e\xe3\x84c\x04l\x04$\xc7{p\x0c\x04\x03\x03\x84\x9f\xe3\x8c#\x0cd\x0c\x84c\x1c;l{l{\xf3n\x9c\xe3\x00\x8c\x0c\xe4\xf2\'\x9c\xdf\x80\x80\x02#\x9c\xe3\xe4\x04\x87\x9c\x1f\x8c\x8c\x84l\xe4c\x0c\xe4#\x84\x8f\x1c#d`\x02\x03\x83\x82\x03lg\x9cp{l\x07\xec\x04l\x9c\x9f\xdc#4 ets_task(40100164, 3, 3fff829c, 4)\r\nOSError: [Errno 2] ENOENT\r\n\r\nMicroPython v1.9.1-8-g7213e78d on 2017-06-12; ESP module with ESP8266\r\nType "help()" for more information.\r\n>>> '
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/rshell/main.py", line 1186, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno 8] nodename nor servname provided, or not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/rshell", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/site-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/usr/local/lib/python3.7/site-packages/rshell/main.py", line 2712, in main
    real_main()
  File "/usr/local/lib/python3.7/site-packages/rshell/main.py", line 2674, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/usr/local/lib/python3.7/site-packages/rshell/main.py", line 1192, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/usr/local/lib/python3.7/site-packages/rshell/main.py", line 1216, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/usr/local/lib/python3.7/site-packages/rshell/main.py", line 1462, in __init__
    Device.__init__(self, pyb)
  File "/usr/local/lib/python3.7/site-packages/rshell/main.py", line 1265, in __init__
    self.has_buffer = self.remote_eval(test_buffer)
  File "/usr/local/lib/python3.7/site-packages/rshell/main.py", line 1379, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "/usr/local/lib/python3.7/site-packages/rshell/main.py", line 1357, in remote
    self.pyb.enter_raw_repl()
  File "/usr/local/lib/python3.7/site-packages/rshell/pyboard.py", line 187, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
rshell.pyboard.PyboardError: could not enter raw repl
```
I tried to investigate why this is happening and it appears that in `pyboard.py` at line #185, you raising a `PyboardError` because you are expecting a response of `raw REPL; CTRL-B to exit` but this board is responding with `Type "help()" for more information.`. (See the end of the long single line of output above the first traceback call).

dhylands, 2018-12-10:
People have reported that they got things to work by adding a pause in pyboard.py
https://github.com/dhylands/rshell/issues/66

I see the stream of gobbly gook that is sent at a different baud rate. I wonder if the commands sent by rshell to enter the raw repl are being sent while the UART is at the weird baud rate?

I guess its time for me to flash one of my ESP8266s with MicroPython and see if I can figure out what's going on.

It would be good to get a trace from wireshark (I know how to do this from linux: see: http://blog.davehylands.com/capturing-usb-serial-using-wireshark/)

It might also mean that rshell needs to split the process of entering the raw REPL into 2 distinct parts: enter the REPL and then once in the REPL enter the raw REPL.

justinmoon, 2019-01-03:
I'm having this same issue. Will try to look into it a little more tomorrow ...