dhylands/rshell #69: [Feature request] rsync exclude option

kevinkk525, 2018-12-22:
Please add the option to exclude files/directories from rsync using the option 
`-exclude images`

peterhinch, 2019-01-12:
I think there is a more flexible way. First to explain to others the use case.

We have a nested directory structure including a project implemented as a Python package. The structure contains files which are to go on GitHub but aren't program files: README's for example. It would be good if users could clone the repo and install to target hardware using rsync, but without filling the target's filesystem with cruft.

I don't relish implementing the Linux rsync approach which is hard to use, let alone to implement or test. I therefore suggest the following "lite" solution.

Each subdirectory optionally may contain a text file `__files__`. If it does not exist, rsync's behaviour is as at present.
If it exists, but is empty, no files from that directory are copied. Subdirectories are treated recursively.
If it exists and contains a list of filenames only files matching that name are copied. Again subdirectories are treated recursively. If a specified file does not exist the fact is silently ignored.

`__files__` must contain only the names of files in the current directory, one filename per line. Directory specifiers are not allowed. `__files__` itself is never copied to the target.

Comments welcome.

As a practical note it may be a while before I get to address this and #70.

kevinkk525, 2019-01-12:
I understand that implementing something as powerful as the linux rsync exclude would be a little too much. 

Using __files__ sounds like a good solution in this context.

dhylands, 2019-01-14:
I would probably lean more towards creating a hidden file, similar in concept to git's .gitignore file, but call it .ignore or .rshell-ignore
This concept could be extended to also allow something like ~/.rshell-ignore which would be like a global filter and a particular directory's .rshell-ignore would be added to that.

peterhinch, 2019-01-17:
That sounds OK, but what about Windows and OSX: do these cope with filenames starting with a `.`?