dhylands/rshell #75: Windows: Arrow keys work in rshell but not it's REPL

gregorypruden, 2019-02-08:
In Windows enter powershell Enter rshell, enter repl, enter a command
>>> import machine
hit up arrow, down error, left arrow, right arrow
>>> HPKM

gregorypruden, 2019-02-08:
After looking at this it is a simple fix.
in getch.py class _GetchWindows replace __call__ the with:

```
    def __call__(self):
        import msvcrt
        ch =  msvcrt.getch()
        if ch == b'\xe0':
            ch = {b'H': b'\x10', b'P': b'\x0e', b'M': b'\x06', b'K': b'\x02'}[msvcrt.getch()]
        return ch 
```

dhylands, 2019-02-09:
Thanks very much for this. I took the essence of what you proposed and recoded it a bit differently in 2cb428e
I also added support for Home/End and DEL keys.

The biggest flaw with the code as you presented it is that if a key is pressed that isn't in the mapping dict then it will cause a KeyError to occur.

closed 2019-02-09 by dhylands

gregorypruden, 2019-02-09:
Glad to help.  I like your solution better because the dict doesn't have a default.  I am pretty new to python and still learning.