dhylands/rshell #77: Whitespace in Windows not supported

aaronnewcomb, 2019-03-10:
I was thinking it would be great to have support in Atom for rshell to be able to quickly upload files to a board right from the IDE without having to use something cumbersome like PlatformIO. However, in my testing, I discovered that rshell isn't able to interpret whitespaces in directory names on Windows. Is there a workaround for this? I couldn't find anything in the documentation. Perhaps it's user error? :)

Here is the output of my command:

C:\>rshell -p COM15 cp /Users/anewcomb/Google Drive/scripts/ESP32MP_Blink/main.py /main.py
Using buffer-size of 32
Connecting to COM15 (buffer-size 32)...
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /boot.py/ /main.py/
Setting time ... Mar 09, 2019 17:30:41
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
Omitting directory /Users/anewcomb/Google
File 'C:\/Drive/scripts/ESP32MP_Blink/main.py' doesn't exist

dhylands, 2019-03-10:
Try enclosing the filename with spaces using single or double quotes.

Otherwise rshell sees 2 arguments (split at the space) rather than a single argument with embedded spaces.

aaronnewcomb, 2019-03-10:
Yeah. I tried quite a few different permutations. Nothing seems to work. The problem, in this case, is with a directory name that has a space ("Google Drive"). I haven't tried a file name with a space yet.

C:\Users\anewcomb\Google Drive>rshell -p COM15 cp \Users\anewcomb\"Google Drive"\scripts\ESP32MP_Blink\main.py /main.py
(...)
File 'C:\Users\anewcomb\Google Drive/\Users\anewcomb"Google' doesn't exist

C:\Users\anewcomb\Google Drive>rshell -p COM15 cp \Users\anewcomb\'Google Drive'\scripts\ESP32MP_Blink\main.py /main.py
(...)
File 'C:\Users\anewcomb\Google Drive/\Users\anewcomb\'Google' doesn't exist

C:\Users\anewcomb\Google Drive>rshell -p COM15 cp /Users/anewcomb/'Google Drive'/scripts/ESP32MP_Blink/main.py /main.py
(...)
File '/Users/anewcomb/'Google' doesn't exist

C:\Users\anewcomb\Google Drive>rshell -p COM15 cp /Users/anewcomb/"Google Drive"/scripts/ESP32MP_Blink/main.py /main.py
(...)
Omitting directory /Users/anewcomb/Google
File 'C:\Users\anewcomb\Google Drive/Drive/scripts/ESP32MP_Blink/main.py' doesn't exist

C:\Users\anewcomb\Google Drive>rshell -p COM15 cp "/Users/anewcomb/Google Drive/scripts/ESP32MP_Blink/main.py" /main.py
(...)
Omitting directory /Users/anewcomb/Google
File 'C:\Users\anewcomb\Google Drive/Drive/scripts/ESP32MP_Blink/main.py' doesn't exist

C:\Users\anewcomb\Google Drive>rshell -p COM15 cp '/Users/anewcomb/Google Drive/scripts/ESP32MP_Blink/main.py' /main.py
(...)
File 'C:\Users\anewcomb\Google Drive/'/Users/anewcomb/Google' doesn't exist

dhylands, 2019-03-11:
I just uploaded 0.0.19 which should support spaces.

aaronnewcomb, 2019-03-11:
Thanks for that! I can confirm that this command did indeed work:

rshell -p COM15 cp '/Users/anewcomb/Google Drive/scripts/ESP32MP_Blink/main.py' /main.py
Using buffer-size of 32
Connecting to COM15 (buffer-size 32)...
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /boot.py/ /main.py/ /webrepl_cfg.py/
Setting time ... Mar 11, 2019 00:32:54
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000