dhylands/rshell #78: Question not an issue about pyboard.py

nbenm, 2019-03-21:
I'm trying to adapt your code to my needs.

If I understand well, here's what you do for executing a command:

1 - enter raw repl
  a - write '\r\x03\x03' # ctrl-C twice: interrupt any running program
  b - read everything
  c - write '\r\x01' # ctrl-A: enter raw REPL
  d - read_until 'raw REPL; CTRL-B to exit\r\n>'
  e - write '\x04' # ctrl-D: soft reset
  f - read_until 'soft reboot\r\n'
  g - read_until 'raw REPL; CTRL-B to exit\r\n'

2 - execute command
  a - read_until '>'
  b - write command
  c - write '\x04'
  d - read 2 bytes and check if its "OK"

3 - exit raw repl
  a - write '\r\x02' # ctrl-B: enter friendly REPL

This doesn't work well for me.

Here's what works fine:

1 - enter raw repl
  a - write '\r\x01' # ctrl-A: enter raw REPL
  b - read_until 'raw REPL; CTRL-B to exit\r\n>'

2 - execute command
  a - write command
  b - write '\x04'
  c - read until "OK"
  d - write '\r\n' -> I don't think it is necessarily this string
  e - read command result until '>'

I don't need to exit raw repl

I can repeat this sequence as many times as I have commands.
I'm I missing something ?

dhylands, 2019-03-21:
The Control-C's are to interrupt a running program. If you know your board is at the REPL, then they aren't needed.

Doing the Control-D (soft-reset) basically reinitializes the heap and puts the board in a known state. If you interrupts a running program then you might not have much free memory left and not be able to do anything.

Entering and staying in the RAW REPL would be a pretty decent optimization.

IIRC pyboard.py collects the output until the end, but splitting the functionality into more granular pieces makes sense.

dhylands, 2019-03-21:
Just to note: I didn't write pyboard,py, it originally came from the micropython/tools directory. pyboard,py is used to run the test suite.

closed 2019-03-23 by nbenm