dhylands/rshell #81: Strange characters are returned by MicroPython to rshell

nbenm, 2019-03-23:
When sending an if sequence example:
```
>>> if (i==1):
...     print ('ok')
... else:
...     print ('ko')
```
after the first print statement, one need to type backspace to remove indentation added automatically. Doing this rshell sends a 0x7f byte to MicroPython.
MicroPython responds 3 bytes 0x1b, 0x5b and 0x4b
I have seen this with a logic analyzer.
Can someone tell what are these three bytes ?
I'm trying to do the same from a calculator, so not a real console, and display stops as soon as these characters come.

My second question is why pressing backspace 0x7F and not 0x08 is sent ?

dhylands, 2019-03-23:
rshell is just passing along the characters it receives. On my linux machine, pressing backspace sends 0x7f, and pressing Control-H sends 0x08. Pressing the DELETE key sends ESC [ 3 ~

The 0x1b 0x5b 0x4b is otherwise known as ESC [ K whiich is the ANSI escape sequence for clear from the cursor to the end of the line without moving the cursor. See: https://en.wikipedia.org/wiki/ANSI_escape_code

These are escape sequences that the console/serial programs understand. rshell doesn't actually do anything with them, it just passes it straight through.

This is where micropython processes the 0x7f (or 127 decimal):
https://github.com/micropython/micropython/blob/e0c6dfe90a3e6265b06265c5b3124c24703776f1/lib/mp-readline/readline.c#L159
This is the function that sends the ESC [ K
https://github.com/micropython/micropython/blob/e0c6dfe90a3e6265b06265c5b3124c24703776f1/lib/mp-readline/readline.c#L83
and its called from here:
https://github.com/micropython/micropython/blob/e0c6dfe90a3e6265b06265c5b3124c24703776f1/lib/mp-readline/readline.c#L340
and its called because the backspace set redraw_from_cursor to true here:
https://github.com/micropython/micropython/blob/e0c6dfe90a3e6265b06265c5b3124c24703776f1/lib/mp-readline/readline.c#L185

So these are all anachronisms of the way serial programs and the consoles work.

dhylands, 2019-03-23:
You'll see that readline.c also has parsers for various escape sequences. to deal with the DELETE. HOME, END, and cursor keys.

You'd probably need to copy that type of logic to add your own parser (except that you'd be parsing different sequences, but the idea is the same).

nbenm, 2019-03-24:
Everything is well explained, thanks.
As I haven't a real console, I'm ignoring these escape sequences.
Backspace characters that are put before do the job.
I'll see what can be done for other keys DELETE, HOME...
For now, the main problem I must solve is that threads don't exist. So handling returned characters, especially when a process takes some time to deliver results, is a headache.

closed 2019-03-24 by nbenm

dhylands, 2019-03-25:
For sending data programmatically, raw more is generally better.