dhylands/rshell #84: unable to access /sd

supersjimmie, 2019-03-30:
I'm trying to use rshell with my esp32, loaded with MicroPython_ESP32_psRAM_LoBo.
Mounting the SDCard works fine:
```
rshell -p /dev/ttyS12 -b 115200 -n
Using buffer-size of 32
Connecting to /dev/ttyS12 (buffer-size 32)...
Trying to connect to REPL  connected
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /flash/
Setting time ... Mar 30, 2019 14:35:19
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 1970
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.
/home/_me_/MicroPython_ESP32_psRAM_LoBo> repl
Entering REPL. Use Control-X to exit.
>
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>>
>>> import uos
>>> uos.sdconfig(uos.SDMODE_1LINE)
>>> uos.mountsd()
---------------------
 Mode:  SD (1bit)
     Name: 00000
     Type: SDSC
    Speed: high speed (50 MHz)
     Size: 971 MB
      CSD: ver=0, sector_size=512, capacity=1989632 read_bl_len=9
      SCR: sd_spec=2, bus_width=5

>>> uos.listdir('/sd')
[]
>>> uos.listdir('/flash')
['aenum.mpy', 'boot.py', 'collections2.mpy', 'examples', 'fonts', 'lib', 'linux_metrics.mpy', 'main.py', 'paho.mpy', 'pint.mpy', 'spiffs.info', 'www']
>>>
```
Then when I return from repl with Ctrl-x, I am able to see /flash, but unable to see /sd:
```
/home/_me_/MicroPython_ESP32_psRAM_LoBo> ls
Documents/         MicroPython_BUILD/ Tools/             liota/
LICENSE.md         README.md          get-pip.py
/home/_me_/MicroPython_ESP32_psRAM_LoBo> ls /flash
aenum.mpy         collections2.mpy  fonts/            linux_metrics.mpy paho.mpy          spiffs.info
boot.py           examples/         lib/              main.py           pint.mpy          www/
/home/_me_/MicroPython_ESP32_psRAM_LoBo> ls /sd
Cannot access '/sd': No such file or directory
```
So I see the local content, I see /flash, but I cannot see /sd.

I tried to restart rshell, but then it fails to load:
```
rshell -p /dev/ttyS12 -b 115200 -n
Using buffer-size of 32
Connecting to /dev/ttyS12 (buffer-size 32)...
Trying to connect to REPL . connected
Testing if ubinascii.unhexlify exists ... b'ets Jun  8 2016 00:22:57\r\n\r\nrst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)\r\nconfigsip: 0, SPIWP:0xee\r\nclk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\nmode:DIO, clock div:2\r\nload:0x3fff0018,len:4\r\nload:0x3fff001c,len:4820\r\nload:0x40078000,len:8764\r\nload:0x40080000,len:5748\r\nentry 0x400802dc\r\n\r\nInternal FS (FatFS): Mounted on partition \'internalfs\' [size: 2162688; Flash address: 0x1F0000]\r\n----------------\r\nFilesystem size: 2101248 B\r\n           Used: 323584 B\r\n           Free: 1777664 B\r\n----------------\r\nMicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32\r\nType "help()" for more information.\r\n>>> '
Traceback (most recent call last):
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/main.py", line 1255, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/_me_/.local/bin/rshell", line 10, in <module>
    sys.exit(main())
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/main.py", line 2843, in main
    real_main()
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/main.py", line 2805, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/main.py", line 1261, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/main.py", line 1285, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/main.py", line 1565, in __init__
    Device.__init__(self, pyb)
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/main.py", line 1339, in __init__
    unhexlify_exists = self.remote_eval(test_unhexlify)
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/main.py", line 1459, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/main.py", line 1437, in remote
    self.pyb.enter_raw_repl()
  File "/home/_me_/.local/lib/python3.5/site-packages/rshell/pyboard.py", line 187, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
rshell.pyboard.PyboardError: could not enter raw repl
```

When I try again, it starts but still no /sd:
```
rshell -p /dev/ttyS12 -b 115200 -n
Using buffer-size of 32
Connecting to /dev/ttyS12 (buffer-size 32)...
Trying to connect to REPL  connected
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /flash/
Setting time ... Mar 30, 2019 14:52:06
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 1970
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.
/home/_me_/MicroPython_ESP32_psRAM_LoBo> ls /flash
aenum.mpy         collections2.mpy  fonts/            linux_metrics.mpy paho.mpy          spiffs.info
boot.py           examples/         lib/              main.py           pint.mpy          www/
/home/_me_/MicroPython_ESP32_psRAM_LoBo> ls /sd
Cannot access '/sd': No such file or directory
/home/_me_/MicroPython_ESP32_psRAM_LoBo>
```

dhylands, 2019-04-01:
This is happening because rshell only scans the root directories once. I'll see about adding a command to re-scan

supersjimmie, 2019-04-02:
I also figured out that this works:
```ls /pyboard/sd```