dhylands/rshell #85: NUCLEO F401RE rshell cp problem

Gamaran, 2019-04-12:
Hi all,

I'm having problem with copying files using rshell. I can copy same file to STM32F4 discovery without problem. But when i try it with NUCLEO F401RE, i got "timed out or error in transfer to remote" error. I tried it with different buffer sizes but no success.

What can be the problem?

dhylands, 2019-04-13:
If you flip the board over there be a sticker that says MB1136 C-02 (where the 02 might be 01 or 03). Which one does your board say?

If it says C-01 then that means its using the internal oscillator by default rather than the 8 MHz MCO clock from the STLINK board. This can cause the baud rates to drift and cause communications errors.

Can you send me the file you're having issues copying? I have an NUCLEO-F401RE (mine is MB1136 C-02) and rshell seems to be working fine for me. You can send the file to dhylands at gmail.com

Gamaran, 2019-04-13:
Thanks for quick reply dhylands.

My board's revision is C-03. All other rshell functions work fine for me also, only cp is the problematic one.

I'll send to you immediately, thanks again

dhylands, 2019-04-15:
I can reproduce the problem here, so I'll investigate and see what I can figure out.

Gamaran, 2019-04-15:
Thanks Dave, I am waiting for your reply.

Dave Hylands <notifications@github.com>, 15 Nis 2019 Pzt, 18:50 tarihinde
şunu yazdı:

yeswearecoding, 2019-04-19:
Hello Dave,

I have the same problem with my NUCLEO-F411RE with STM32F411xE board.
My board revision is C-04.

I hope it will be the same part of code.

dhylands, 2019-06-19:
Please try rshell 0.0.26 

I just uplaoded a fix. Please let me know if it wrks for you.

Gamaran, 2019-07-15:
Hi Dave, 

rshell 0.0.26 solves my problem. cp command works without problem now.

Thank you very much for your help.

closed 2019-07-15 by Gamaran