dhylands/rshell #91: rshell improvement with rts / dtr

camelator, 2019-04-26:
Hi,
I think my issue is related to this post:
https://forum.micropython.org/viewtopic.php?p=35345
trying to connect to MaixPy does not work because of RTS signal detected by firmware and the firmware jumps to example .
The root cause are DTR / RTS signal , set to high in the constructor of class Serial.
By changing the code in file pyboard.py (according to miniterm program) it works for me.
Here is what I did:
                        self.serial = serial.Serial(device, baudrate=baudrate, inter_byte_timeout=1,rtscts=True,dsrdtr=True)
                        self.serial.rts=False
                        self.serial.dtr=False 

Unfortunately I am not a python developper.
What I suggest is to add a command line parameter or a cfg file to change the values according to user needs (like miniterm. It is a good example). Maybe someone can submit some python code?

dhylands, 2019-04-26:
Which MaixPy do you have? And perhaps, which OS are you using? Under linux, I'm not having any issues connecting to my MaixPy (the Maix Bit).

dhylands, 2019-04-26:
Looking at this a bit further:
```
self.serial = serial.Serial(device, baudrate=baudrate, inter_byte_timeout=1,rtscts=True,dsrdtr=True)
self.serial.rts=False
self.serial.dtr=False
```
I don't think that we want to pass rtscts=True and dsrdtr=True as these enable hardware flow control which we definitely don't want. Looking at the schematics it appears that RTS and DTR are connected thru similar circuitry as is used by the ESP32/8266 to control BOOT & RESET. So you don't want those toggling during normal transfers (which might happen when flow control is enabled).

The two lines that set rts and dtr to False seem like the useful addtions (although I'd be inclined to add them as command line options).

dhylands, 2019-04-27:
I added --rts and --dtr command line options (and you can also set the defaults via RSHELL_RTS or RSHELL_DTR environment variables.

dhylands, 2019-04-27:
This is included in rshell v0.0.23

camelator, 2019-04-27:
Many thanks :) you are right about rtscts and dsrdtr. Only the last two lines are important. I am running rshell on Ubuntu 18.04.
Many thanks for the update!

closed 2019-05-24 by dhylands