dhylands/rshell #93: Extraneous characters displayed at Windows shell

klaff, 2019-04-29:
Using rshell (in 32-bit Windows 7), the console appears to not handle the Esc ]K sequence returned by the pyboard after a backspace is sent.  For example, if one enters "type" followed by a backspace, the python prompt deletes the "e", but also displays "←[K". Each additional backspace continues the pattern.

```
>>> type
<class 'type'>
>>> typ←[K
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'typ' is not defined
```

dhylands, 2019-04-30:
This is actually an artifact of the console. MicroPython is sending the ESC [ K sequence and rshell is just passing it on to the console, and the console doesn't know what to do with it. Linux, Mac, and Windows 10 consoles know hoe to deal with this escape sequence.

I supposed rshell could have a command line option to have it detect these sequences and do something else, I'd just have to figure out what the something else is.

klaff, 2019-04-30:
Thanks. It does seem to be only a Windows 7 issue and we have found that by using the alternate shell https://github.com/cmderdev/cmder that we can work around the problem.