dhylands/rshell #103: Correct usage of rsync? Bricked host system

elixx, 2019-05-18:
I had used the rsync command successfull a few times, to sync a webserver directory to an ESP32 board. Until now..
I assumed the DEST_DIR argument was relative to the target device, but that doesn't appear to be the case. I was trying to sync to the local "deploy/" directory to the root of the device, "/" and... it ran crazy on my local "/":

```
┌─[root@localhost] - [/mnt/tmp/espserver] - [2019-05-18 01:34:21]
└─[0] <git:(dev 814e8dc✗✱✈) > rshell -p /dev/ttyUSB0
Using buffer-size of 32
Connecting to /dev/ttyUSB0 (buffer-size 32)...
Trying to open serial port /dev/ttyUSB0
Serial port opened successfully True
Trying to connect to REPL  connected
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /utils.py/ /config.py/ /esp32thing.py/ /microWebSocket.py/ /microWebTemplate.py/ /www/ /microWebSrv.py/ /boot.py/ /webrepl_cfg.py/ /main.py/ /client.cfg/
Setting time ... May 18, 2019 13:34:23
Evaluating board_name ... pyboard
Retrieving time epoch ... rJan 01, 2000
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.
/mnt/tmp/espserver> rsync deploy/ /
/mnt/tmp/espserver> rsync
usage: rsync [-m|--mirror] [-n|--dry-run] [-q|--quiet] SRC_DIR DEST_DIR
rsync: error: the following arguments are required: SRC_DIR, DEST_DIR
/mnt/tmp/espserver> rsync -m deploy/ /
Removing //snap
Removing //run
Removing //sbin
Removing //media
Removing //root

^C

^C^C^C^CRemoving //etc
Removing //srv
Removing //usr
^C^C^C^C^C^Z
[1]  + 6103 suspended  rshell -p /dev/ttyUSB0
mygit:1: command not found: git
┌─[root@localhost] - [/mnt/tmp/espserver] - [2019-05-18 05:35:17]
└─[0] <> ls -AlF /
total 188
drwxr-xr-x   2    0    0  4096 May  9 09:19 bin/
drwxr-xr-x   5    0    0  4096 Oct 24  2018 boot/
-rw-r--r--   1    0    0   234 May 18 17:30 boot.py
-rw-r--r--   1    0    0    13 May 18 17:30 client.cfg
-rw-r--r--   1    0    0    13 May 18 17:30 client.disable
-rw-r--r--   1    0    0   185 May 18 17:30 config.py
drwxr-xr-x  16    0    0  7660 May 18 16:54 dev/
drwxr-xr-x  22    0    0 12288 May 18 17:35 etc/
drwxr-xr-x   2    0    0  4096 Nov 16  2018 home/
drwxr-xr-x  20    0    0  4096 Oct 25  2018 lib/
drwx------   2    0    0 16384 Oct 24  2018 lost+found/
-rw-r--r--   1    0    0    73 May 18 17:30 main.py
-rw-r--r--   1    0    0 10795 May 18 17:30 microWebSocket.py
-rw-r--r--   1    0    0 35414 May 18 17:30 microWebSrv.py
-rw-r--r--   1    0    0 13261 May 18 17:30 microWebTemplate.py
drwxr-xr-x   5    0    0  4096 Oct 30  2018 mnt/
drwxr-xr-x   3    0    0  4096 Apr 25 08:04 opt/
dr-xr-xr-x 310    0    0     0 Jan  1  1970 proc/
-rw-r--r--   1 1001 1002    62 Sep 29  2018 README.txt
lrwxrwxrwx   1    0    0    17 Oct 25  2018 root -> /mnt/sd/home/root/
drwxr-xr-x  13    0    0   380 May 18 17:34 run/
drwxr-xr-x   2    0    0 12288 May 18 17:34 sbin/
lrwxrwxrwx   1    0    0     7 Oct 26  2018 sd -> /mnt/sd/
dr-xr-xr-x  12    0    0     0 May 15 12:11 sys/
drwxrwxrwt  15    0    0 16384 May 18 16:32 tmp/
drwxr-xr-x  11    0    0  4096 Dec  5 12:03 usr/
-rw-r--r--   1    0    0  4155 May 18 17:30 utils.py
drwxr-xr-x  17    0    0  4096 Nov  5  2018 var/
drwxr-xr-x   4    0    0  4096 May 18 17:30 www/
mygit:1: command not found: git
┌─[root@localhost] - [/mnt/tmp/espserver] - [2019-05-18 05:35:21]
└─[0] <> ls /snap
ls: cannot access '/snap': No such file or directory
```

Of course, the system didn't come up after a reboot...

elixx, 2019-05-18:
This is really painful. I should have used the dry-run option.

elixx, 2019-05-18:
Another missed best practice; `chown /dev/ttyUSB0` and don't run as root. :-/

dhylands, 2019-05-19:
The source and fest directories can either be for the host or the target.

This is described in the opening paragraphs of the documentation:
https://github.com/dhylands/rshell#rshell

I normally setup udev rules, or add your user to the dialout group so that
you don't need to run as root.

elixx, 2019-05-20:
I somehow missed the bits about "/flash" and "/sdcard". Not sure how it ever worked for me, without that!

Thanks for the pointer. My miss.

closed 2019-05-20 by elixx