dhylands/rshell #104: Verbose mode not documented

cefn, 2019-05-25:
There seems to be a `--verbose (-v)` option to the rsync command but it's not detailed in the documentation excepting in an oblique reference via `--dry-run`.

cefn, 2019-05-26:
OK, this is a lot more nuanced than I thought. 

There doesn't seem to be a verbose flag, but there appeared to be, because somehow quiet=True is ending up as the default for my rsync call, even though the default for quiet is documented as `False`. 

Also surprisingly, this unexpected quietness is resolved by adding a `-v` flag to rsync even there doesn't appear to be such a flag at https://github.com/dhylands/rshell/blob/master/rshell/main.py

Very surprising. Any thoughts?

# Example

I begin the session by wiping an ESP8266 altogether (so rsync has a clean start). There is no vgkits/console folder in the 0.1.0 vanguard+rainbow build, so that's what I will be rsyncing on the board...

```
cefn-bionic-thinkpad:console$ vanguard brainwash
Running 'esptool.py --port /dev/ttyUSB0 erase_flash'
[...]
Running 'esptool.py --port /dev/ttyUSB0 --baud 1500000 write_flash --flash_mode dout --flash_size 1MB 0 /home/cefn/.local/lib/python3.6/site-packages/vgkits/vanguard/data/firmware/vanguard+rainbow-0.1.0.bin'
[...]
Wrote 1048576 bytes (423700 compressed) at 0x00000000 in 7.2 seconds (effective 1168.2 kbit/s)...
[...]
cefn-bionic-thinkpad:console$ ampy --port /dev/ttyUSB0 mkdir --exists-okay vgkits
cefn-bionic-thinkpad:console$ ampy --port /dev/ttyUSB0 mkdir vgkits/console
```

Then I run rsync without the `-v` flag...

```
cefn-bionic-thinkpad:console$ rshell --ascii --buffer-size=30 --port /dev/ttyUSB0 rsync ./ /pyboard/vgkits/console
Connecting to /dev/ttyUSB0 ...
cefn-bionic-thinkpad:console$
```

Unfortunately rshell doesn't report anything copied as you can see. It appears to have copied it, but did this silently, even though I didn't specify the quiet flag.

If I repeat the flash procedure but instead run the rsync command with a -v flag I get what I would expect from a non-quiet execution...

```
cefn-bionic-thinkpad:console$ rshell --ascii --buffer-size=30 --port /dev/ttyUSB0 rsync -v ./ /pyboard/vgkits/console
Connecting to /dev/ttyUSB0 ...
Adding /pyboard/vgkits/console/oldconsole.py
Adding /pyboard/vgkits/console/mainMenu.py
Adding /pyboard/vgkits/console/testRun.py
Adding /pyboard/vgkits/console/webGameConsole.py
Adding /pyboard/vgkits/console/async
Adding /pyboard/vgkits/console/async/asyncConsole.py
Adding /pyboard/vgkits/console/gameConsole.py
Adding /pyboard/vgkits/console/examples
Adding /pyboard/vgkits/console/examples/chat.py
Adding /pyboard/vgkits/console/examples/yura.py
Adding /pyboard/vgkits/console/examples/mathdice.py
Adding /pyboard/vgkits/console/examples/hello.py
Adding /pyboard/vgkits/console/examples/molly.py
Adding /pyboard/vgkits/console/examples/hangman.py
Adding /pyboard/vgkits/console/examples/consequences.py
Adding /pyboard/vgkits/console/examples/menu.py
```

dhylands, 2019-05-27:
rsync prior to 0.0.22 had a verbose flag and quiet mode was the default. From 0.0.22 onwards, rsync has a --quiet flag and verbose mode is the default.