dhylands/rshell #107: Windows: Arrow keys still not working in repl

wokibe, 2019-06-15:
The problem should be solved after #75 has been patched with 2cb428e, but I still can not use the arrow keys in the repl mode of rshell.

The Windows version is 1803 (OS Build 17134.829)

(esp) C:\Users\wolfkibe\Projects\micropython>python -V
Python 3.7.2

And rshell is the actual 0.0.25
(esp) C:\Users\wolfkibe\Projects\micropython>pip list
Package    Version
---------- -------
ecdsa      0.13.2
esptool    2.6
pip        19.1.1
pyaes      1.6.1
pyreadline 2.1
pyserial   3.4
pyudev     0.21.0
rshell     0.0.25
setuptools 40.6.2
six        1.12.0

I am sure, that the _Getch.__init__ recognized this as Windows installation (as I  inserted a print statement after the self.impl = _GetchWindows()), but I was not successful in finding, why the arrow keys are not recognized.

Best regards
Wolfgang Kittenberger
(wolfkibe@gmail.com)

wokibe, 2019-06-18:
Further investigations show, that the builtin repl does not recognize the _GetchWindows  substitutions of the self.keymap. Instead I tried to simplify the corresponding __call__ method:
    
    class _GetchWindows():
        def __init__(self):
            import msvcrt    # just to recognize a windows installation

        def __call__(self):
            import sys
            ch = sys.stdin.buffer.read(1)
            return ch

Now the arrow keys, home, end, insert, delete can be used in the repl started within rshell.

Best regards
Wolfgang Kittenberger

wokibe, 2019-07-01:
The suggested change did not really solve the problem, as the ...read(1) does not supress the echo. So  I need to continue to use msvcrt, but change the handling of the control characters. Here the proposed code:

    class _GetchWindows:
        def __init__(self):
            import msvcrt
            import queue
            # Windows sends 0xe0 followed by a single letter for many of the
            # special keys (like left/right arrow)
            # so we map these to the characters MicroPythons 'repl' expects
            self.keymap = {
                b'K': '[D',   # Left Arrow
                b'S': '[3~',  # DEL
                b'M': '[C',   # Right Arrow
                b'P': '[B',   # Down Arrow (Next line in history)
                b'H': '[A',   # Up Arrow (Prev line in history)
            }

            # as the expected control sequences have some bytes, we need a reservoir
            self.rsv = queue.SimpleQueue()

        def __call__(self):
            import msvcrt
            if not self.rsv.empty():
                ch = self.rsv.get().encode('utf_8')
                return ch
            ch = msvcrt.getch()
            if ch == b'\xe0':
                ch = msvcrt.getch()
                if ch in self.keymap:
                    # fill the reservoir
                    for chr in self.keymap[ch]:
                        self.rsv.put(chr)
                    # start  the control sequence
                    ch = b'\x1b'
            return ch

Sorry, I have tried to generate a pull request, but failed.
Best regards
Wolfgang Kittenberger

dhylands, 2019-07-02:
Is this under Windows 7?

The arrow keys seem to be working fine for me under WIndows 10.

wokibe, 2019-07-02:
I can confirm, this is a Windows 10 installation:

C:\Users\wolfkibe>systeminfo

Host Name:                 DESKTOP-Q5L6S6K
OS Name:                   Microsoft Windows 10 Home
OS Version:                10.0.17134 N/A Build 17134
OS Manufacturer:           Microsoft Corporation
OS Configuration:          Standalone Workstation
OS Build Type:             Multiprocessor Free
Registered Owner:          N/A
Registered Organization:   N/A
Product ID:                00325-96272-16705-AAOEM
Original Install Date:     07/03/2019, 22:11:05
System Boot Time:          20/06/2019, 23:12:56
System Manufacturer:       ASUSTeK COMPUTER INC.
System Model:              FX503VM
System Type:               x64-based PC
Processor(s):              1 Processor(s) Installed.
                           [01]: Intel64 Family 6 Model 158 Stepping 9 GenuineIntel ~2800 Mhz
BIOS Version:              American Megatrends Inc. FX503VM.311, 29/04/2019
Windows Directory:         C:\WINDOWS
System Directory:          C:\WINDOWS\system32
Boot Device:               \Device\HarddiskVolume1
System Locale:             en-gb;English (United Kingdom)
Input Locale:              de;German (Germany)
Time Zone:                 (UTC+01:00) Amsterdam, Berlin, Bern, Rome, Stockholm, Vienna
Total Physical Memory:     8.148 MB
Available Physical Memory: 4.342 MB
Virtual Memory: Max Size:  9.428 MB
Virtual Memory: Available: 2.697 MB
Virtual Memory: In Use:    6.731 MB
Page File Location(s):     C:\pagefile.sys
Domain:                    WORKGROUP
Logon Server:              \\DESKTOP-Q5L6S6K
Hotfix(s):                 8 Hotfix(s) Installed.
                           [01]: KB4100347
                           [02]: KB4230204
                           [03]: KB4485449
                           [04]: KB4493478
                           [05]: KB4497398
                           [06]: KB4497932
                           [07]: KB4503308
                           [08]: KB4503286
Network Card(s):           4 NIC(s) Installed.
                           [01]: Realtek PCIe GBE Family Controller
                                 Connection Name: Ethernet
                                 DHCP Enabled:    Yes
                                 DHCP Server:     192.168.2.1
                                 IP address(es)
                                 [01]: 192.168.2.39
                                 [02]: fe80::a5fa:ff02:8f19:a175
                           [02]: Intel(R) Dual Band Wireless-AC 8265
                                 Connection Name: WiFi
                                 Status:          Media disconnected
                           [03]: Bluetooth Device (Personal Area Network)
                                 Connection Name: Bluetooth Network Connection
                                 Status:          Media disconnected
                           [04]: VirtualBox Host-Only Ethernet Adapter
                                 Connection Name: VirtualBox Host-Only Network
                                 DHCP Enabled:    No
                                 IP address(es)
                                 [01]: 192.168.56.1
                                 [02]: fe80::6157:59ff:4808:8f2
Hyper-V Requirements:      VM Monitor Mode Extensions: Yes
                           Virtualization Enabled In Firmware: Yes
                           Second Level Address Translation: Yes
                           Data Execution Prevention Available: Yes

The only thing, wich will be different to your installation is the "Input Locale" set to "de;German". But I dont think, that this has influence on the arrows control keys.

Best regards
Wolfgang Kittenberger

dhylands, 2019-07-02:
Which shell are you using to launch rshell?

My system is running a slightly newer version of Windows 10 - build 17763
I tested using cmd.exe
My input locale is set to en-us;English (United States)

wokibe, 2019-07-03:
Yes, I am also using cmd.exe
Anyway, I think we are looking on the wrong place.

In the comments to _GetchWindows you write:

        import msvcrt
        # Windows sends 0xe0 followed by a single letter for many of the
        # special keys (like left/right arrow) so we map these to the
        # characters that MicroPythons readline function will work with.

as msvcrt delivers the expected characters, the challenge lies in the MicroPythons rwadline function. So it is not a Windows issue, but related to the ESP8266.

I am using a LoLin NodeMCU development board V3 (that one, that is to broad!).
Here is the output of the flashing:

     (esp) C:\Users\wolfkibe\Projects\micropython>python esptool.py write_flash 0 esp8266-20190529-v1.11.bin
  esptool.py v2.6
  Found 1 serial ports
  Serial port COM4
  Connecting....
  Detecting chip type... ESP8266
  Chip is ESP8266EX
  Features: WiFi
  MAC: 84:f3:eb:e3:35:d8
  Uploading stub...
  Running stub...
  Stub running...
  Configuring flash size...
  Auto-detected Flash size: 4MB
  Flash params set to 0x0040
  Compressed 617880 bytes to 402086...
  Wrote 617880 bytes (402086 compressed) at 0x00000000 in 35.6 seconds (effective 139.0 kbit/s)...
  Hash of data verified.
  Leaving...
  Hard resetting via RTS pin...
  (esp) C:\Users\wolfkibe\Projects\micropython>

So the micropython version is esp8266-20190529-v1.11.bin from micropython.org/download

Best regards
Wolfgang Kittenberger

Nick2253, 2020-04-17:
I am also having this problem with Windows 10, both Powershell and cmd.  And I am also using a NodeMCU v3 board.

Were you able to identify a fix, or will we have to wait for changes in the MicroPython firmware for ESP8266?