dhylands/rshell #109: rshell chaining commands issue when ending ~ is present

ramprao, 2019-07-12:
`MicroPython v1.11-145-gb574edf21-dirty on 2019-07-10; ESP32 module with ESP32` 

I have a weird issue when running some programs directly from the command line using rshell. 

Here is my code:
```
$ cat f0.py
print("Begin")

N = 150000
# Simple loop
for i in range(N):
    pass

print("End")
```

`$ rshell -p /dev/ttyUSB0 repl \~ import f0 \~`

This does not seem to  fully execute the script. 

```
Setting time ... Jul 11, 2019 14:17:51
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
Entering REPL. Use Control-X to exit.
>
MicroPython v1.11-145-gb574edf21-dirty on 2019-07-10; ESP32 module with ESP32
Type "help()" for more information.
>>>
>>> import f0
Begin
```
I get back my prompt. The same program runs completely when I stay in the REPL after the import.

`$ rshell -p /dev/ttyUSB0 repl \~ import f0`
```
Setting time ... Jul 11, 2019 14:21:34
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
Entering REPL. Use Control-X to exit.
>
MicroPython v1.11-145-gb574edf21-dirty on 2019-07-10; ESP32 module with ESP32
Type "help()" for more information.
>>>
>>> import f0
Begin
End
>>>
```
These programs seem to take about 2+ seconds or so. If I reduce the value of N, I can get the program to fully execute even with the ~  at the end.

timhughes, 2020-11-03:
I am seeing a similar issue. I think this is because it executes the command and doesn't wait for the output. Having some way to stay attached while the repl is running and then detaching at the end would be great for running tests on real hardware. 

I see a similar thing when i try and run my unit tests
```
] $ rshell -p /dev/ttyUSB0 repl "~ from test.test_server import *~ unittest.main()~ print('done')~ "
Using buffer-size of 32
Connecting to /dev/ttyUSB0 (buffer-size 32)...
Trying to connect to REPL  connected
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /lib/ /tinyweb/ /test/
Setting time ... Nov 03, 2020 19:06:39
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
Entering REPL. Use Control-X to exit.
>
MicroPython v1.13 on 2020-09-02; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> from test.test_server import *; unittest.main(); print('done')
```

```
] $ rshell -p /dev/ttyUSB0 repl "~ from test.test_server import *~ unittest.main()~ print('done') "
Using buffer-size of 32
Connecting to /dev/ttyUSB0 (buffer-size 32)...
Trying to connect to REPL  connected
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /lib/ /tinyweb/ /test/
Setting time ... Nov 03, 2020 19:06:47
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
Entering REPL. Use Control-X to exit.
>
MicroPython v1.13 on 2020-09-02; ESP32 module with ESP32
Type "help()" for more information.
>>> 
>>> from test.test_server import *; unittest.main(); print('done')
testUrldecode (Utils) ... ok
testParseQueryString (Utils) ... ok
testRequestLine (ServerParts) ... ok
testRequestLineEmptyLinesBefore (ServerParts) ... ok
testRequestLineNegative (ServerParts) ... ok
testHeadersSimple (ServerParts) ... ok
testHeadersSpaces (ServerParts) ... ok
testHeadersEmptyValue (ServerParts) ... ok
testHeadersMultiple (ServerParts) ... ok
testUrlFinderExplicit (ServerParts) ... ok
testUrlFinderParameterized (ServerParts) ... ok
testUrlFinderNegative (ServerParts) ... ok
testRouteDecorator1 (ServerFull) ... ok
testRouteDecorator2 (ServerFull) ... ok
testResourceDecorator1 (ServerFull) ... ok
testResourceDecorator2 (ServerFull) ... ok
testStartHTML (ServerFull) ... ok
testRedirect (ServerFull) ... ok
testRequestBodyUnknownType (ServerFull) ... ok
testRequestBodyJson (ServerFull) ... ok
testRequestBodyUrlencoded (ServerFull) ... ok
testRequestBodyNegative (ServerFull) ... ok
testRequestLargeBody (ServerFull) ... ok
testRouteParameterized (ServerFull) ... ok
testParseHeadersOnOff (ServerFull) ... ok
testDisallowedMethod (ServerFull) ... ok
testAutoOptionsMethod (ServerFull) ... ok
testPageNotFound (ServerFull) ... ok
testMalformedRequest (ServerFull) ... ok
testOptions (ServerResource) ... ok
testGet (ServerResource) ... ok
testGetWithParam (ServerResource) ... ok
testGetWithArgs (ServerResource) ... ok
testGenerator (ServerResource) ... ok
testPost (ServerResource) ... ok
testInvalidMethod (ServerResource) ... ok
testException (ServerResource) ...ERROR: something
 ok
testBrokenPipe (ServerResource) ... ok
testSendFileManual (StaticContent) ... ok
testSendFileNotFound (StaticContent) ... ok
testSendFileConnectionReset (StaticContent) ... ok
Ran 41 tests

OK
>>> 
```

timhughes, 2020-11-03:
I have worked around this using pyboard 

```pyboard --device /dev/ttyUSB0 -b 115200 -c 'from test.test_server import *; unittest.main();'```