dhylands/rshell #111: RFC Macro facility

peterhinch, 2019-08-31:
This would add a facility to import a Python module containing a `dict` of macros. At the **rshell** prompt a macro would be expanded. For example, a user macro `copy` taking a filename argument would be expanded and run as follows:
```
> m copy myfile
> cp myfile /path/to/my/project/
```
To include a user specified macro file **rshell** would be invoked with
```
$ rshell -m my_macro_file
```
which would cause it to import `my_macro_file.py` by searching the Python path in the usual way.

A typical macro file might look like
```python
from my_global_macros import macros  # Optional
macros['copy'] = 'cp {} /path/to/my/project\n', 'Copy a file to LED flashing project.'
```
A macro module exports a `dict` named `macros` whose key is the macro name. The value is either a string or a 2-tuple. In the latter case element[0] is the expansion and element[1] is a help string. The latter would be accessed by the `ml` (macro list) command
```
> ml copy
```
This would display the expansion string and the help string (if present). Issued with no args, `ml` would list all available macros.

Note that arguments are supported by Python string `format` syntax so there can be any number with arbitrary placement in the expansion.

### Design

The macro facility is likely to be used by experienced programmers maintaining larger projects. The proposal therefore assumes competence in Python and prioritises flexibility over user friendliness.

There are potential issues over recursive macros and ones whose expansion contains newline characters. I would propose disallowing recursion and allowing a newline only as the last character. If a newline is present in that position the macro will auto-execute, otherwise it will be expanded but only run if the user presses `Enter`.

Macros would only be expanded at the `>` prompt. At the `$` prompt the user would surely employ scripting.

Lastly this is based on functionality I would find useful. I have not yet studied how to implement this :)

Comments welcome, in particular whether anyone else would find this useful.

peterhinch, 2019-09-03:
This can be tried via [my fork](https://github.com/peterhinch/rshell.git).

dhylands, 2019-09-04:
I think that the macro facility is a great idea.

peterhinch, 2019-09-05:
In that case I'll submit a PR.

I'm already finding it useful, notably in projects using Python packages with a nested directory structure and long rshell commands.

The solution to the [mv](https://github.com/dhylands/rshell/issues/46) command would be safer if there were a fix to https://github.com/dhylands/rshell/issues/112.