dhylands/rshell #114: Regression: rshell V0.0.26 doesn't work with FTDI adaptor

peterhinch, 2019-09-27:
The FTDI connection works with miniterm.py. Test conditions:
```
[adminpete@capybara]: /usr/local/bin
$ ./rshell --version
0.0.26
```
When invoked it hangs at this point:
```
[adminpete@capybara]: /usr/local/bin
$ ./rshell -p /dev/ftdi --buffer-size=30 -a
Using buffer-size of 30
Connecting to /dev/ftdi (buffer-size 30)...
Trying to connect to REPL . connected
Testing if ubinascii.unhexlify exists ...
```
If interrupted when hanging the traceback is as follows:
```
[adminpete@capybara]: /usr/local/bin
$ ./rshell -p /dev/ftdi --buffer-size=30 -a
Using buffer-size of 30
Connecting to /dev/ftdi (buffer-size 30)...
Trying to connect to REPL . connected
Testing if ubinascii.unhexlify exists ... ^CTraceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1282, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./rshell", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/dist-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 2913, in main
    real_main()
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 2875, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1288, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1312, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1602, in __init__
    Device.__init__(self, pyb)
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1366, in __init__
    unhexlify_exists = self.remote_eval(test_unhexlify)
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1490, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "/usr/local/lib/python3.6/dist-packages/rshell/main.py", line 1468, in remote
    self.pyb.enter_raw_repl()
  File "/usr/local/lib/python3.6/dist-packages/rshell/pyboard.py", line 206, in enter_raw_repl
    data = self.read_until(1, b'raw REPL; CTRL-B to exit\r\n')
  File "/usr/local/lib/python3.6/dist-packages/rshell/pyboard.py", line 164, in read_until
    data = self.serial.read(min_num_bytes)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt
[adminpete@capybara]: /usr/local/bin
```

bpjobin, 2019-11-08:
I'm not sure if I'm having the exact same issue but it looks like it.
I'm trying to connect to an esp8266 (feather huzzah). I tried using version 0.0.21 to 26. Using it on a mac.

```
Bruno:~/ $ rshell --port /dev/tty.SLAB_USBtoUART --buffer-size=30 -a                                                      [20:18:34]
Using buffer-size of 30
Connecting to /dev/tty.SLAB_USBtoUART (buffer-size 30)...
Trying to connect to REPL .
```
It juste hang there forever. 
I'm not having this problem with an esp32.

bpjobin, 2019-11-08:
I should probably mention that this board in flashed with circuitpython v3.2.1.
I'm not having this issue with a nodeMCU running micropython.