dhylands/rshell #115: Can I use rshell to SSH into my ESP32 via the built in WiFi.

jdcrunchman, 2019-10-02:


jdcrunchman, 2019-10-02:
I’m trying to find an SSH client that would allow me to SSH into my ESP32 over a network, using the built in WiFi support on the ESP32.

I found rshell!  If I install it on my ESP32, then set my IP address to the wlan then use a normal SSH client I have running on my iPad so I can access the esp python shell via SSh?

Please contact me via email at jdcrunchman at gmail dot com

John

dhylands, 2019-10-02:
rshell doesn't know anything about ssh.

ASIDE: I'm not sure exactly what you expect ssh'ing into an ESP32 would even do. Normally ssh would run a shell (i.e. bash, csh etc), and your ssh session gives you access to that shell. THe ESP32 doesn't have any "shells" other than the REPL.

I suppose that if ssh were supported, you might be able to ssh in and get to a REPL. webrepl does something similar to this without the secure aspect.

closed 2019-10-02 by dhylands

jdcrunchman, 2019-10-03:
Sorry my bad,   I meant SFTP,   but I think Ampy can do that.

John