dhylands/rshell #116: rsync copies 2 or 3 files then failes with: timed out or error in transfer to remote

reillybeacom, 2019-10-06:
sorry if this is a duplicate

dhylands, 2019-10-07:
Can you provide more details?
What is your host computer?
What micropython board are you uising?
What version of rshell are you using?

surlymatt, 2020-02-04:
I have same issue on Windows 10.
Running in python 3.7 in pipenv.
Using command prompt (not powershell)
Connect via usb to a genuine ESP32-DevKitC
Maybe one or two files copy, but then it just hangs.
No time outs, no error messages. Nothing.
Device must be physically disconnected to release the com port, then start all over again.
Was replicated on multiple PCs by various users.
Target hardware was running this frmware: esp32-idf3-20191220-v1.12.bin