dhylands/rshell #117: WSL ESP32: cp/rsync to board: timed out or error in transfer to remote

scy, 2019-10-29:
Hi there!

I’m trying to copy or rsync files using rshell, but receive the error message `timed out or error in transfer to remote`.

* Doing `ls /pyboard` works, the REPL works too.
* Copying files via ampy works as well.
* Copying files _from_ the board (to the host) works too, just not the other way around.
* The board is a ESP32, my host is running Windows 10 with Ubuntu in WSL 1.
* The board is connected via USB serial, but WSL translates this to `/dev/ttyS3` instead of `/dev/ttyUSB3`.
* The error occurs both with and without `-a` and both with the default buffer size of 32 as well as 512. Adding `--wait 3` doesn’t help either.
* This is rshell 0.0.26 and MicroPython 1.11.

Failing commands:

* `rshell -p /dev/ttyS3 'rsync --mirror pycode /pyboard'`
* `cd pycode && rshell -p /dev/ttyS3 'rsync --mirror . /pyboard'`
* `cd pycode && rshell -p /dev/ttyS3 'cp config.py /pyboard'`

These fail regardless of whether I pass the command to rshell on the command line or enter them interactively.

Since the error message doesn’t tell a lot, I’ve hacked `~/.local/lib/python3.6/site-packages/rshell/main.py` to change the error message line to:

```python
sys.stderr.write("timed out or error in transfer to remote: {0}\n".format(ack))
```

This way, I figured out that instead of `0x06` (ACK, acknowledge), the board is responding with `0x04` (EOT, end of transmission).

This is the output of `stty -F /dev/sttyS3 -a`:

```
speed 115200 baud; rows 0; columns 0; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^U; eof = <undef>; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z; rprnt = ^R; werase = ^W; lnext = ^V; discard = ^O; min = 1; time = 10;
-parenb -parodd -cmspar cs8 hupcl -cstopb cread clocal -crtscts
-ignbrk -brkint -ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl -onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon -iexten -echo -echoe -echok -echonl -noflsh -xcase -tostop -echoprt -echoctl -echoke -flusho -extproc
```

Any suggestions?

jarritto1, 2020-02-04:
Did you ever resolve this? I have the same problem.

scy, 2020-02-04:
@jarritto1, I did not. Still waiting for response by @dhylands or someone. My workaround was to use ampy instead. Which has other drawbacks, of course.

jarritto1, 2020-02-05:
Unfortunately for me ampy has been even less reliable with transferring files.  I suppose I'll stick to using rshell rsync in a loop until it completes without error messages.

jarritto1, 2020-02-21:
Just wanted to give an update on rsync.  I've been using rsync from the command line instead of inside rshell and my success rate so far has been 100%.  (i.e. rshell -b 115200 -p COM5 rsync ./ /pyboard)

PR #130 (2020-04-13): Be more patient in send_file_to_remote
I was experiencing many failures writing files to my PyBoard until I increased the timeout in `send_file_to_remote` from 1 s to 2 s. I have not had a timeout error since.

This should also resolve #117, assuming the WiPy experiences similar flash write times.
closed 2020-04-13 by dhylands