dhylands/rshell #119: Linux Mint 19 install errors

uxhamby, 2019-11-09:
bjh@laptop2:~$ sudo pip3 install rshell
The directory '/home/bjh/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/bjh/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Collecting rshell
  Downloading https://files.pythonhosted.org/packages/11/e6/1cf4129a4d49eb4ea5007d9f292fa22a8ba9d2b533c179b9df9ae3d51dd2/rshell-0.0.26.tar.gz (44kB)
    100% |████████████████████████████████| 51kB 269kB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-n3dd3_fd/rshell/
bjh@laptop2:~$ sudo -H pip3 install rshell
Collecting rshell
  Using cached https://files.pythonhosted.org/packages/11/e6/1cf4129a4d49eb4ea5007d9f292fa22a8ba9d2b533c179b9df9ae3d51dd2/rshell-0.0.26.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-6csgqmk2/rshell/
bjh@laptop2:~$ rshell

Command 'rshell' not found, did you mean:

davehylands, 2019-11-10:
It looks like your installation is missing setuptools.

Try this:
```
sudo apt-get install python3-setuptools
```

uxhamby, 2019-11-11:
That seems to have done the trick. Thanks!