dhylands/rshell #122: AttributeError: module 'serial' has no attribute 'VERSION'

pacmac, 2019-12-25:
Just did. fresh install on Debian, but cannot get it to run, any ideas ??

```
Connecting to /dev/ttyUSB0 (buffer-size 32)...
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/rshell/main.py", line 1282, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/rshell", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/usr/local/lib/python3.7/dist-packages/rshell/main.py", line 2913, in main
    real_main()
  File "/usr/local/lib/python3.7/dist-packages/rshell/main.py", line 2875, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/usr/local/lib/python3.7/dist-packages/rshell/main.py", line 1288, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/usr/local/lib/python3.7/dist-packages/rshell/main.py", line 1312, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/usr/local/lib/python3.7/dist-packages/rshell/main.py", line 1554, in __init__
    pyb = Pyboard(port, baudrate=baud, wait=wait, rts=RTS, dtr=DTR)
  File "/usr/local/lib/python3.7/dist-packages/rshell/pyboard.py", line 130, in __init__
    if serial.VERSION == '3.0':
AttributeError: module 'serial' has no attribute 'VERSION'
```

davehylands, 2019-12-27:
It sounds like you have an old version of pyserial.

I would start by trying:
```
python3 -m pip install --upgrade pyserial
```
You'll probably also need to add `sudo`

electricWah, 2022-08-06:
I am getting this exact same error, with all the newest versions of everything. After running `pip install --upgrade rshell`:
```
Requirement already satisfied: rshell in ./Library/Python/3.8/lib/python/site-packages (0.0.31)
Requirement already satisfied: pyudev>=0.16 in ./Library/Python/3.8/lib/python/site-packages (from rshell) (0.23.2)
Requirement already satisfied: pyserial in ./Library/Python/3.8/lib/python/site-packages (from rshell) (3.5)
Requirement already satisfied: six in /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/site-packages (from pyudev>=0.16->rshell) (1.15.0)
```

krishnak, 2023-07-19:
sudo pip3 install --ignore-installed -U pyserial

Run this command to get it resolved