dhylands/rshell #133: rsync is always seeing all the files as newer

PureBasic, 2020-05-23:
[image]
I am missing something here. Does the ESP32 need to have an RTC for this to work?

PureBasic, 2020-05-25:
Alright this is not happening all the time. I'm keeping this open and will post more info if I discover a pattern.

AltoRetrato, 2020-07-15:
@tionebrr , what file system are you using? FAT or LittleFS?

tionebrr, 2020-07-16:
@AltoRetrato , FAT.

scy, 2020-09-04:
I’m having a similar problem on my ESP32. However, when starting rshell, I’ve noticed that it does something like this:

```
Setting time ... Sep 05, 2020 00:46:22
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
```

So apparently the time is not set correctly, which of course will lead to all files on the board being from Jan 01 2000.

@tionebrr, does the same thing happen to you?

scy, 2020-09-05:
Okay, my problem went away after upgrading to the new MicroPython 1.13 release. Before, I was using an 1.12-based nightly build from August in order to get uasyncio support. Looks like it had a somewhat incorrect filesystem implementation.

Since you didn’t tell us the version of MicroPython you were using, maybe you’re seeing the same effect.

tionebrr, 2020-09-05:
Okay.
I'm not using micropython these time so I think I'll close the issue.
Thanks for the feedbacks @scy .

closed 2020-09-05 by tionebrr