dhylands/rshell #137: Confusion about timezones

aivarannamaa, 2020-08-31:
I'm adding time synchronization to Thonny IDE and while looking for prior work, I first turned to your exellent tool. In main.py I found useful information in `set_time`, but also following confusing piece:

```
        # The pyboard maintains its time as localtime, whereas unix and
        # esp32 maintain their time as GMT
        self.adjust_for_timezone = (epoch_tuple[0] != 1970)
```

I have a Pyboard lite 1.0 with latest stable firmware (1.12). Whenever I write a file, `os.stat(file_name)[8]` (ie. ST_MTIME) becomes equal to `time.localtime()`. As the time fields in the stat structure are supposed to be in UTC, it looks like Pyboard also maintains its time as UTC/GMT. (I fail to see how it could be anything else, as Pyboard API does not have any timezone related properties.)

While some firmwares (eg. Pycom's) do have timezone support, the `RTC().init` still seems to expect UTC time as input, so why is the adjustment required at all?

Even if it is required, I don't understand why you make it dependent on the kind of epoch the device is using.

davehylands, 2020-09-01:
For the most part, micropython has no notion of timezones (so localtime is the same as gmtime). So whatever time you initialize the RTC to is the same time that's used to timestamp files. For most people, this is local time. VFAT filesytems also typically use localtime for their timestamp.

Versions of micropython that have network support typically use some form of NTP to maintain time, and since NTP is UTC based with a 1970 epoch, that's what gets used to initialize the RTC.

For the pyboard, time is localtime and has an epoch of 2000.

Using the epoch is a fairly reliable way to distinguish whether the RTC is using UTC or localtime.

Here's a summary:
pyboard - epoch 2000 - RTC is localtime
unix - epoch 1970 - doesn't have RTC but file timestamps are all UTC
ESP series - epoch 1970 - RTC is UTC
pyboard D - not sure

aivarannamaa, 2020-09-18:
Thank you for this comprehensive answer and sorry for the delay in answering! I took time to argue with you, but now when I'm rereading your post, I don't want to argue anymore :)

One thing doesn't look correct, though:

> ESP series - epoch 1970

See
```
MicroPython v1.12 on 2019-12-20; ESP module with ESP8266
Type "help()" for more information.
>>> import time
>>> time.localtime(0)
(2000, 1, 1, 0, 0, 0, 5, 1)
```
and

```
MicroPython v1.13 on 2020-09-02; ESP32 module with ESP32
Type "help()" for more information.
>>> import time
>>> time.localtime(0)
(2000, 1, 1, 0, 0, 0, 5, 1)
```

aivarannamaa, 2020-09-18:
I still couldn't help and raised this topic also in MicroPython forum: https://forum.micropython.org/viewtopic.php?f=2&t=9047

aivarannamaa, 2020-09-22:
The issue is much clearer now. Thank you!

closed 2020-09-22 by aivarannamaa