dhylands/rshell #138: rshell fails on M5stack

dclaar, 2020-09-10:
I'm trying to connect to an M5stack Atom Mini. I am using 0.0.28 on windows 10. I get "could not enter raw repl". I'm wondering if it is due to the M5Stack's cutsie logo?

I get:

```
rshell -pcom9
Using buffer-size of 32
Connecting to com9 (buffer-size 32)...
Trying to connect to REPL  connected
Testing if ubinascii.unhexlify exists ... b'\r\n       _  __ _               \r\n _   _(_)/ _| | _____      __\r\n| | | | | |_| |/ _ \\ \\ /\\ / /\r\n| |_| | |  _| | (_) \\ V  V / \r\n \\__,_|_|_| |_|\\___/ \\_/\\_/  \r\n\r\nAPIKEY: `NUMBER`\r\n'
Traceback (most recent call last):
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\main.py", line 1307, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno 11001] getaddrinfo failed
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\Scripts\rshell-script.py", line 11, in <module>
    load_entry_point('rshell==0.0.28', 'console_scripts', 'rshell')()
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\command_line.py", line 4, in main
    rshell.main.main()
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\main.py", line 2966, in main
    real_main()
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\main.py", line 2928, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\main.py", line 1313, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\main.py", line 1337, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\main.py", line 1627, in __init__
    Device.__init__(self, pyb)
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\main.py", line 1391, in __init__
    unhexlify_exists = self.remote_eval(test_unhexlify)
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\main.py", line 1515, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\main.py", line 1493, in remote
    self.pyb.enter_raw_repl()
  File "C:\Users\dougm\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\rshell\pyboard.py", line 209, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
rshell.pyboard.PyboardError: could not enter raw repl
```

If I connect with putty, I get the following. Note that it actually stops at `APIKEY` until I hit control-x, then it goes into the cutsie UIFLOW ascii logo and the rest.
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5116
load:0x40078000,len:10596
load:0x40080400,len:5744
entry 0x400806c0

       _  __ _
 _   _(_)/ _| | _____      __
| | | | | |_| |/ _ \ \ /\ / /
| |_| | |  _| | (_) \ V  V /
 \__,_|_|_| |_|\___/ \_/\_/

APIKEY: `NUMBER`
Traceback (most recent call last):
  File "_flow.py", line 12, in <module>
  File "wifiCfg.py", line 105, in auto_connect
  File "wifiCfg.py", line 94, in connect_ui
KeyboardInterrupt:
MicroPython 400ff3bbf-dirty on 2020-08-06; M5Stack with ESP32
Type "help()" for more information.
>>>
>>>
>>> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> configsip: 188777542, SPIWP:0xee
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> mode:DIO, clock div:1
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> load:0x3fff0018,len:4
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> load:0x3fff001c,len:5116
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> load:0x40078000,len:10596
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> load:0x40080400,len:5744
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> entry 0x400806c0
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>>
>>>        _  __ _
Traceback (most recent call last):
  File "<stdin>", line 1
IndentationError: unexpected indent
>>>  _   _(_)/ _| | _____      __
Traceback (most recent call last):
  File "<stdin>", line 1
IndentationError: unexpected indent
>>> | | | | | |_| |/ _ \ \ /\ / /
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> | |_| | |  _| | (_) \ V  V /
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>>  \__,_|_|_| |_|\___/ \_/\_/
Traceback (most recent call last):
  File "<stdin>", line 1
IndentationError: unexpected indent
>>>
>>> APIKEY: `NUMBER`
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```

davehylands, 2020-09-15:
If you use putty, try the following:
* Press Control-C until you get to the prompt
* Press Control-D

What does the console output look like?

dclaar, 2020-09-20:
Did it a few times, got this:
```
KeyboardInterrupt:
MicroPython 400ff3bbf-dirty on 2020-08-06; M5Stack with ESP32
Type "help()" for more information.
>>>
MPY: soft reboot

       _  __ _
 _   _(_)/ _| | _____      __
| | | | | |_| |/ _ \ \ /\ / /
| |_| | |  _| | (_) \ V  V /
 \__,_|_|_| |_|\___/ \_/\_/

APIKEY: KEY
Traceback (most recent call last):
  File "_flow.py", line 12, in <module>
  File "wifiCfg.py", line 105, in auto_connect
  File "wifiCfg.py", line 94, in connect_ui
KeyboardInterrupt:
MicroPython 400ff3bbf-dirty on 2020-08-06; M5Stack with ESP32
Type "help()" for more information.
>>>
MPY: soft reboot

       _  __ _
 _   _(_)/ _| | _____      __
| | | | | |_| |/ _ \ \ /\ / /
| |_| | |  _| | (_) \ V  V /
 \__,_|_|_| |_|\___/ \_/\_/

APIKEY: KEY
```

One time I did it a bit fast, and got:

```
KeyboardInterrupt:
MicroPython 400ff3bbf-dirty on 2020-08-06; M5Stack with ESP32
Type "help()" for more information.
>>> MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>>
>>>        _  __ _
Traceback (most recent call last):
  File "<stdin>", line 1
IndentationError: unexpected indent
>>>  _   _(_)/ _| | _____      __
Traceback (most recent call last):
  File "<stdin>", line 1
IndentationError: unexpected indent
>>> | | | | | |_| |/ _ \ \ /\ / /
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> | |_| | |  _| | (_) \ V  V /
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>>  \__,_|_|_| |_|\___/ \_/\_/
Traceback (most recent call last):
  File "<stdin>", line 1
IndentationError: unexpected indent
>>>
>>> APIKEY: KEY
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>>
```

Which looks a lot like what I see with rshell. Wondering if it is a timing thing.

davehylands, 2020-09-21:
It looks like this is some type of bug in the M5Stack where it's taking output (perhaps from the previous run) and reading it as input?

rshell definitely won't work properly if the soft-reset comes up with all of those syntax errors (i.e. it doesn't arrive at a clean prompt)

ropg, 2021-05-24:
Please have a look at https://github.com/dhylands/rshell/issues/27 and see if trying with my experimental `-s`, `--suppress-reset` command-line option solves it.

ropg, 2021-05-24:
Ha, I have M5Stack Core2, flashed it with UiFlow and it indeed did not work without the -s option, and works fine with it.

```text
cf:~/code $ rshell -n
Using buffer-size of 32
Connecting to /dev/cu.usbserial-01FC0209 (buffer-size 32)...
Trying to connect to REPL  connected
Retrieving sysname ... b'\r\n       _  __ _               \r\n _   _(_)/ _| | _____      __\r\n| | | | | |_| |/ _ \\ \\ /\\ / /\r\n| |_| | |  _| | (_) \\ V  V / \r\n \\__,_|_|_| |_|\\___/ \\_/\\_/  \r\n\r\nassertion "ATB_GET_KIND(block) == AT_HEAD" failed: file "../../py/gc.c", line 590, function: gc_free\r\nabort() was called at PC 0x401e6bc7 on core 1\r\n\r\nELF file SHA256: 0000000000000000\r\n\r\nBacktrace: 0x4009a1df:0x3ffcf760 0x4009a5d1:0x3ffcf780 0x401e6bc7:0x3ffcf7a0 0x400d367d:0x3ffcf7d0 0x400d2d45:0x3ffcf7f0 0x4013a293:0x3ffcf810 0x40137749:0x3ffcf830 0x40121772:0x3ffcf850 0x4012180f:0x3ffcf870 0x40123697:0x3ffcf890 0x401317b1:0x3ffcf8c0 0x40173a5b:0x3ffcf8f0 0x400e4719:0x3ffcf910 0x400e048d:0x3ffcf930 0x400e05de:0x3ffcf950 0x400e6128:0x3ffcf970 0x400e6163:0x3ffcf9a0 0x400ebab9:0x3ffcf9c0 0x400e0662:0x3ffcfaa0 0x400eeebd:0x3ffcfae0 0x400e48f2:0x3ffcfb80 0x400e048d:0x3ffcfbf0 0x400e04b6:0x3ffcfc10 0x400eb8d9:0x3ffcfc30 0x400eb956:0x3ffcfcc0 0x400ebccd:0x3ffcfcf0 0x400e0662:0x3ffcfdd0 0x400eeebd:0x3ffcfe10 0x400e48f2:0x3ffcfeb0 0x400e048d:0x3ffcfee0 0x400e04b6:0x3ffcff00 0x40175678:0x3ffcff20 0x40175941:0x3ffcffb0 0x400f82c6:0x3ffcffe0 0x40095009:0x3ffd0010\r\n\r\nRebooting...\r\nets Jul 29 2019 12:21:46\r\n\r\nrst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)\r\nconfigsip: 0, SPIWP:0xee\r\nclk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\nmode:DIO, clock div:1\r\nload:0x3fff0018,len:4\r\nload:0x3fff001c,len:5280\r\nho 0 tail 12 room 4\r\nload:0x40078000,len:13424\r\nload:0x40080400,len:3568\r\nentry 0x4008063c\r\n\r\n       _  __ _               \r\n _   _(_)/ _| | _____      __\r\n| | | | | |_| |/ _ \\ \\ /\\ / /\r\n| |_| | |  _| | (_) \\ V  V / \r\n \\__,_|_|_| |_|\\___/ \\_/\\_/  \r\n\r\nAPIKEY: 92F4B270\r\nMicroPython e876aac88-dirty on 2021-04-29; M5Stack with ESP32\r\nType "help()" for more information.\r\n>>> '
Traceback (most recent call last):
  File "/Users/rop/.espressif/python_env/idf4.0_py3.9_env/bin/rshell", line 33, in <module>
    sys.exit(load_entry_point('rshell', 'console_scripts', 'rshell')())
  File "/Users/rop/code/esp/rshell/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/Users/rop/code/esp/rshell/rshell/main.py", line 3099, in main
    real_main()
  File "/Users/rop/code/esp/rshell/rshell/main.py", line 3061, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/Users/rop/code/esp/rshell/rshell/main.py", line 1391, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/Users/rop/code/esp/rshell/rshell/main.py", line 1415, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/Users/rop/code/esp/rshell/rshell/main.py", line 1747, in __init__
    Device.__init__(self, pyb)
  File "/Users/rop/code/esp/rshell/rshell/main.py", line 1499, in __init__
    self.sysname = self.remote_eval(sysname)
  File "/Users/rop/code/esp/rshell/rshell/main.py", line 1634, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "/Users/rop/code/esp/rshell/rshell/main.py", line 1612, in remote
    self.pyb.enter_raw_repl()
  File "/Users/rop/code/esp/rshell/rshell/pyboard.py", line 212, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
rshell.pyboard.PyboardError: could not enter raw repl
cf:~/code $ rshell -n -s
Using buffer-size of 32
Connecting to /dev/cu.usbserial-01FC0209 (buffer-size 32)...
Trying to connect to REPL  connected
Retrieving sysname ... esp32
Testing if ubinascii.unhexlify exists ... Y
Retrieving root directories ... /flash/
Setting time ... May 24, 2021 09:28:54
Evaluating board_name ... pyboard
Retrieving time epoch ... Jan 01, 2000
Welcome to rshell. Use Control-D (or the exit command) to exit rshell.
/Users/rop/code> ls /pyboard/flash
Wi-Fi-Config.py apps/           blocks/         boot.py         img/            main.py         res/            temp.py         update/        
/Users/rop/code> 
```

(The -n is because something is weird (sometimes) about the terminal when I use colors, ignore that for now...)

davehylands, 2021-05-24:
Based on the output here, I'd say that there is some type of corruption on your filesystem.

ropg, 2021-05-25:
It's a fresh copy of whatever M5Stack gives you, on top of 'make erase'

More generally: there are lots of reasons for micropython to cause errors on soft-boot that cause it to escalate to hard-reset. I don't think many people test that whatever they flash cleanly unloads for the soft reboot. For example the ili9341 pure python driver does not cleanly unload, and there are many other possible reasons, including someone developing something that doesn't unload properly yet. In all these cases rshell / pyboard (IMHO) need to be resistant to that and be able go to raw REPL in cases where soft-reset does not put them back there.

(Alternatively, pyboard should detect that a hard reset has happened and do a Ctrl-A again. But then issuing multiple commands takes forever because it waits for the boot every time, and it's not easy to do if/then with send/expect logic.)

`-s` is the easiest way to help people stuck at `could not enter raw repl`

ropg, 2021-05-25:
Additionally, you might want access to the Python REPL in the state that the thing is in currently, not in the state after a soft-reboot.