dhylands/rshell #139: Rshell crash on MacOS BigSur

zhakhalov, 2021-01-14:
Hello,

Since installation of MacOS Big Sur `rshell` command fails with the following error
```
rshell -p /dev/tty.SLAB_USBtoUART
Traceback (most recent call last):
  File "/usr/local/bin/rshell", line 5, in <module>
    from rshell.command_line import main
  File "/usr/local/lib/python3.8/site-packages/rshell/command_line.py", line 1, in <module>
    import rshell.main
  File "/usr/local/lib/python3.8/site-packages/rshell/main.py", line 52, in <module>
    from serial.tools import list_ports
  File "/usr/local/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module>
    from serial.tools.list_ports_posix import comports
  File "/usr/local/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module>
    from serial.tools.list_ports_osx import comports
  File "/usr/local/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
```

davehylands, 2021-01-14:
That looks like an issue with pyserial. The issue was raised here:
https://github.com/pyserial/pyserial/issues/509
and fixed here:
https://github.com/pyserial/pyserial/pull/542

So if you update your pyserial to the latest (or at least v3.5), then this problem should go away.

zhakhalov, 2021-01-17:
Thanks, @davehylands. It helps.

closed 2021-01-17 by zhakhalov