dhylands/rshell #142: How to name the board.  Typo in the README.rst

Naohiro2g, 2021-01-27:
I made a board.py file in the rshell as documented in the README.rst;

echo name="myboard" > /pyboard/board.py

then exit rshell once.

When I reconnect to my esp32 board, rshell had the error message like below.

> Evaluating board_name ... pyboard
----- Prints from board.py ----
Error encountered executing board.py
Traceback (most recent call last):
  File "<stdin>", line 4, in board_name
  File "board.py", line 1, in <module>
NameError: name 'myboard' isn't defined

I found the board.py contains name=myboard instead of name="myboard", so the message was correct.

I could change the name of board like below;
echo 'name="myboard"' > /pyboard/board.py

davehylands, 2021-01-28:
Thanks. Fixed in #1bf8d12ee43a410675073d2b020db6d4b0d6f32b

closed 2021-01-28 by davehylands

Naohiro2g, 2021-01-28:
Thank YOU.
Because RP2040, Pico, and others have come,  rshell and MIcroPython should be in the limelight, soon.

samveen, 2023-01-16:
@davehylands something related to document is that a space is required between the redirection `>` and other parameters , unlike regular shells. It's probably the lowest possible priority, but would help save a tiny bit of time.