dhylands/rshell #145: rshell --help isn't very helpful

fake-name, 2021-02-04:
Effectively the title. The help output of rshell doesn't actually list the commands that rshell supports, making it less then helpful.

```
durr@tinybox ~/P/E/workSpace (master)> rshell --help
usage: rshell [options] [command]

Remote Shell for a MicroPython board.

positional arguments:
  cmd                   Optional command to execute

optional arguments:
  -h, --help            show this help message and exit
  -b BAUD, --baud BAUD  Set the baudrate used (default = 115200)
  --buffer-size BUFFER_SIZE
                        Set the buffer size used for transfers (default = 512 for USB, 32 for UART)
  -p PORT, --port PORT  Set the serial port to use (default 'None')
  --rts RTS             Set the RTS state (default '')
  --dtr DTR             Set the DTR state (default '')
  -u USER, --user USER  Set username to use (default 'micro')
  -w PASSWORD, --password PASSWORD
                        Set password to use (default 'python')
  -e EDITOR, --editor EDITOR
                        Set the editor to use (default 'vi')
  -f FILENAME, --file FILENAME
                        Specifies a file of commands to process.
  -d, --debug           Enable debug features
  -n, --nocolor         Turn off colorized output
  -l, --list            Display serial ports
  -a, --ascii           ASCII encode binary files for transfer
  --wait WAIT           Seconds to wait for serial port
  --timing              Print timing information about each command
  -V, --version         Reports the version and exits.
  --quiet               Turns off some output (useful for testing)

You can specify the default serial port using the RSHELL_PORT environment variable.

```

So from the above, `cmd` seems to be the placeholder for \*things\* that make `rshell` do stuff. What are the commands supported? Who knows!

This isn't even a case of "well, it's _actually_ documented in the man pages", since those don't seem to exist.
(believe me, that's another argument since some platforms remove the man pages for space reasons).

```
durr@tinybox ~/P/E/workSpace (master)> man rshell
No manual entry for rshell
```

Entering the `rshell` cli (why do I need a CLI in my CLI?), apparently you can type `help` (hmmm, maybe that's one of the commands?), and get something useful:

```

/home/durr/Programming/ESPScreen/workSpace> help
Executing "help"

Documented commands (type help <topic>):
========================================
args    cat  connect  date  edit  filesize  help  mkdir  rm     shell
boards  cd   cp       echo  exit  filetype  ls    repl   rsync

````

Is there a reason that's not output when `rshell --help` is invoked? Instead of

```
positional arguments:
  cmd                   Optional command to execute
```

also, apparently `rshell help` is _different_ then `rshell --help`. Wat.

misan, 2021-05-05:
>also, apparently rshell help is different then rshell --help. Wat.

The former is a command and the latter is an option, 
as in 

> usage: rshell [options] [command]

davehylands, 2021-05-12:
rhsell --help shows the command line options available when invoking rshell (and using --help is the standard way that most command line tools do this).

When you invoke rshell you can also pass commands to be executed and `help` is one of those commands. If you start rshell and then issue the `help` command it shows the available commands from within rshell and this is what happens when you run `rshell help`. You could also run `rshell ls .` to get a listing of files in the current directory.

It doesn't make sense for `rshell help` to show all of the command line options that show up in --help since none of those are applicable once you've started rshell. It might make sense for `rshell --help` to also include the output of the `help` command.